KiCad/kicad-mac-builder #11: Setup the output paths of the DMGs

adamwolf, 2018-02-27:
Right now, I think the output paths of the DMGs are hardcoded.  If so, fix that, otherwise close this issue.

nickoe, 2018-03-26:
A bit related to this, the dmgs get put into build/package-kicad/src/dmg/ ... that is a strange path. They are in a src folder?

The dmg are named for example `kicad-20180326-170338.abc-KMB.dmg`

I guess that includes the date and time. abc looks like a placeholder for a git SHA? What is KMB supposed to mean?

adamwolf, 2018-03-27:
Agreed.  Before this issue gets closed, nightlies, extras, and
unified/stable dmgs need to have sensible names in a good place.

KMB was for Kicad Mac builder, and it was just a temporary thing.  Abc was
the placeholder for a SHA.

I am not beholden to any of it.

If anyone here had good suggestions for naming scheme, let's talk it out,
or else just submit a PR.

I want to get all these issues burned down so I can take out "it doesn't
work yet" from the description :)

nickoe, 2018-03-27:
I don't think we need KMB in the name, but lets discuss that in a another issue. This is not so important yet. But I have started to push the dmg's to http://downloads.kicad-pcb.org/osx such that they are accessible to anyone who would like to test.

adamwolf, 2018-03-29:
Thanks @craftyjon!  We still need to do the same change for extras, and then we can close this out.

closed 2018-03-30 by adamwolf