KiCad/kicad-mac-builder #12: Setup DMG READMEs

adamwolf, 2018-02-27:
The DMGs need usage READMEs that are valid for the new system.

adamwolf, 2018-04-26:
They need to say "You must open KiCad first before opening the standalone apps, or else the standalone apps won't be able to open up due to macOS quarantining."

adamwolf, 2018-04-27:
You can model from the current READMEs over at https://github.com/wayneandlayne/KiCadMacOSPackaging

danngreen, 2018-05-29:
README might also want to say something about how to upgrade from an older version when using a case-sensitive drive.
Dragging the "KiCad" folder (uppercase 'C') on the DMG to the Applications folder in order to overwrite a previous folder named "Kicad" (lowercase 'c'), will give an error if the user has a case-insensitive driver (which is the default in OSX).

The OSX error dialog doesn't have a button to allow overwriting the "Kicad" folder with the "KiCad" folder, so the user has to manually remove/rename the old folder.

Screenshot of the error dialog attached:
[image]

PR #148 (2018-06-22): Internal README and packaging updates.
Fixes #12
closed 2018-06-22 by adamwolf