KiCad/kicad-mac-builder #17: OSX deployment target mismatch

nickoe, 2018-02-27:
```
[...]
-- The CXX compiler identification is AppleClang 7.3.0.7030031
CMake Error at /Users/kicad/homebrew/Cellar/cmake/3.6.2/share/cmake/Modules/Platform/Darwin.cmake:84 (message):
  CMAKE_OSX_DEPLOYMENT_TARGET (10.13) is greater than CMAKE_OSX_SYSROOT SDK:

   /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk

  Please set CMAKE_OSX_DEPLOYMENT_TARGET to 10.11 or lower.
Call Stack (most recent call first):
  /Users/kicad/homebrew/Cellar/cmake/3.6.2/share/cmake/Modules/CMakeSystemSpecificInformation.cmake:36 (include)
  CMakeLists.txt:32 (project)

-- Configuring incomplete, errors occurred!
```
And I get this version info from the host:
```
$ sw_vers -productVersion
10.11.6
```
What is the expected behaivour?  I see the TODO mentions that it has to be tested on 10.11, I guess this is the result?

adamwolf, 2018-02-27:
I'll take a look.  It should work on 10.11 through 10.13.

Thanks for the bug report.

adamwolf, 2018-02-28:
Could you test https://github.com/wayneandlayne/kicad-mac-builder/pull/19?

adamwolf, 2018-02-28:
I built just fine in a 10.11 VM, so I merged the fix.  If you could make a build from master and let me know, I'll mark this closed.

adamwolf, 2018-03-26:
Due to not setting MACOS_MIN_VERSION, fixing with https://github.com/wayneandlayne/kicad-mac-builder/pull/42.

closed 2018-03-26 by adamwolf