KiCad/kicad-mac-builder #4: There exists a build.sh but there are manual instructions int he readme

nickoe, 2018-02-27:
In the readme instructions to call generic cmake and make commands are presented, but yet there is a `build.sh` script in the source tree. What is the way it is supposed to be used on a CI?

adamwolf, 2018-02-27:
In my CI, I use build.sh through a vagrant box.  This lets me use one CI
machine to build it inside of a 10.11 box, a 10.12 box, and a 10.13 box,
and the machines are "fresh" every time, so nothing from a build can stick
around.

When I'm on business travel, I like to build locally as often I do not have
internet.  I'll add a task to create a ci.sh to make it clearer, but
there's already a task to make the readme readable to other people.

adamwolf, 2018-02-27:
Working on making this clearer on the https://github.com/wayneandlayne/kicad-mac-builder/tree/readme branch.

adamwolf, 2018-03-05:
Closing after tweaking readme.

closed 2018-03-05 by adamwolf