KiCad/kicad-mac-builder #5: Minimum CMake version requirement

nickoe, 2018-02-27:
Is there any reason to require cmake 3.8? Can 3.6.2 be used?

adamwolf, 2018-02-27:
I have no idea.  I use CLion frequently, and it uses 3.8.  If this is
something you want to help out on, I can definitely look into it real soon,
otherwise I'll put investigating this into the list of tasks for after this
is ready for v5.

nickoe, 2018-02-27:
It is simply because my box has 3.6.2. I will test it when I get home to see if it works with earlier and may propose to change this.

adamwolf, 2018-02-27:
Awesome!  I bet it works fine on 3.6.2, and if not, I have no problem with
making it work on that.

nickoe, 2018-02-27:
It looks like it works fine with 3.6.2, but I have not yet made a successful build becuase of other issues that I will report.

adamwolf, 2018-02-28:
Pulled it back to 3.6.2.  Closing.

closed 2018-02-28 by adamwolf