KiCad/kicad-mac-builder #9: Review wxPython issue

adamwolf, 2018-02-27:
There is a longstanding issue with wxpython.  If folks have it already installed, KiCad doesn't work quite right.

This *may* be fixed by the new Python approach of this installer.  If not, investigate if we can fix it by changing the pythonpath of the embedded Python.

adamwolf, 2018-04-27:
This is fixed for me, but I want to get proof from someone else before I close it out.

adamwolf, 2018-05-02:
As soon as #86 gets into nick's nightlies, we can test this and put this bug to bed.

adamwolf, 2018-05-04:
Hi nick,

I pushed some changes that may fix this.  Could you remind me which version of cmake you're using on that system?  BundleUtilities looks a little different than what I have here.

nickoe, 2018-05-04:
Ok, new issue, it can't find wx.
http://ci.kicad-pcb.org/job/osx-kicad-adam-new-head/109/console

adamwolf, 2018-05-04:
nah, it can find wx fine, the actual error is there: 08:21:28 error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: for: /Volumes/SSD/workspace/osx-kicad-adam-new-head/build/python-dest/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python (for architecture x86_64) option "-add_rpath @executable_path/../Frameworks" would duplicate path, file already has LC_RPATH for: @executable_path/../Frameworks
08:21:28 make[3]: *** [python/src/python-stamp/python-fixup] Error 1
08:21:28 make[2]: *** [CMakeFiles/python.dir/all] Error 2
08:21:28 make[2]: *** Waiting for unfinished jobs....

and that actually brings me to the next point.

As far as I can tell, that means that the python-dest directory is being reused between jobs.  Normally that would be fine, but the kicad patching I've been doing over the last week means we need to clear out that workspace.  Is that something you can do before tonight's build?  (I have a patch that makesadding rpaths more gentle and not die if they're already there, but that won't fix the state of the KiCad source directory.)

adamwolf, 2018-05-05:
Thanks so much for your patience, nick.  We're *so* close!  I'm running into an issue now where there's some difference in... something tricky between 10.11 and 10.13. Not sure I'll get the solution tonight, but I'm trying!

adamwolf, 2018-05-05:
As of current master, I did a completely clean build, and an incremental build, on two different 10.11 machines and both succeeded.  Let's see what happens on the KiCad build machine...

nickoe, 2018-05-07:
I get some errors related to six and I did try to make a completely clean build:
```
00:00:23.759 make[3]: *** No rule to make target `six/src/six-stamp/six-skip-update', needed by `six/src/six-stamp/six-configure'.  Stop.
```
http://ci.kicad-pcb.org/job/osx-kicad-adam-new-head/116/console

adamwolf, 2018-05-07:
I'll take a look at the six issues.  I spent the weekend setting up a
public Jenkins with clean VMs I can build on to try to get this solved
faster.

I'm also seeing a weird issue where it's finding your brew's bison
directory, but it also says your bison is too old.  Any chance you can
check which version is in BISON_PATH?

Thanks a bunch, nick!

adamwolf, 2018-05-09:
nick, any ideas why I'm building fine on a clean 10.11 VM, clean 10.12 VM, and on a 10.13 dev machine, and yet it isn't building on the c4osx box?  I can dig into it today but if you have any insight please let me know.  If you want to look at the Jenkins logs, an example success from last night is https://builder.wayneandlayne.com/job/kicad-mac-builder-10.11/job/master/9/

adamwolf, 2018-05-09:
I wonder if it could be differences in GetPrerequisites.cmake from 3.6.2 to 3.11.1.  If not, I wonder if it's a change in otool itself?

nickoe, 2018-05-09:
I currently have no idea, and I have not yet had tine to try to create an updated brew install.  I will not be able to make any progress here before next week. I may attempt to compare the log before, but only if we are lucky.

adamwolf, 2018-05-09:
No problem.  I have time to dig into it.  Could you maybe set it to build
twice a day while I tweak some things?

nickoe, 2018-05-09:
I set it to 9 and 12, although I am not sure of the timezone. But it only built at 9 before, I hope that matches sensibly up with your timezone.

adamwolf, 2018-05-09:
I reverted my dev box to the older cmake, so we will see if I can reproduce
it.  I am optimistic, and the solution there might be as easy as improving
some new cmake files in the Kicad repo.

adamwolf, 2018-05-10:
Awesome!  When I use cmake 3.8.2, I get the same problem!  I'm going to make a new bug for this.

adamwolf, 2018-05-18:
If someone could check this on the newest builds from http://downloads.kicad-pcb.org/osx/testing/, it would be awesome.

adamwolf, 2018-05-20:
Confirmed fixed as of 20180519 nickoe builds.

closed 2018-05-20 by adamwolf