KiCad/kicad-mac-builder #21: Confirm KICAD_SYS_SYMBOL_DIR is set correctly

adamwolf, 2018-03-02:
Today I noticed it was set to /Library/Application/library which is definitely not right.

holvo, 2018-06-09:
This happend to me with an atual Windows nightly:

It is caused by an old sym-lib-table found in
C:\Users\holger\AppData\Roaming\kicad\sym-lib-table.

It dated from 22.5.2017, and all path entries contain KICAD_SYS_SYMBOL_DIR
But KICAD_SYS_SYMBOL_DIR is outdated, KICAD_SYMBOL_DIR is the way to go.

The old sym-lib-table with KICAD_SYS_SYMBOL_DIR  was left over from a previous KiCad installation and has not been overwritten by todays new installation.

Only after deleting it and calling Eeschema, I was asked to select a new table.
The the selected default global symbol libtary is o.k..

adamwolf, 2018-06-22:
Thanks!  I haven't seen this except that one time, and the dev list is already working on creating -lib-table migration instructions.

closed 2018-06-22 by adamwolf