KiCad/kicad-mac-builder #23: make wxpython download fails

nickoe, 2018-03-04:
See http://ci.kicad-pcb.org/job/osx-kicad-adam-new-head/13/console

adamwolf, 2018-03-04:
Weird.  Works for me! :)  I'll do a clean and see what's going on.

nickoe, 2018-03-04:
When I wget the path I also get 404. It looks like the wxpython project does not exist on sf.net anymore right now. Maybe a workaround would be to use the wxpython source package downloaded form the debian archives or something like that.

adamwolf, 2018-03-05:
Hmm.  I think it must sometimes be pointing to mirrors that don't have the file.  I poked Sourceforge folks to see if this is typical.

nickoe, 2018-03-26:
I suspect this was intended to be fixed by https://github.com/wayneandlayne/kicad-mac-builder/commit/76dba92263e1892a65b8d100d6007881e8995381

But it does not seem to work for me, I get:
```
00:03:06.207 [ 68%] Performing download step (verify and extract) for 'wxpython'
00:03:06.212 cd /Volumes/SSD/workspace/osx-kicad-adam-new-head/kicad-mac-builder/build/wxpython-prefix/src && /Users/kicad/homebrew/Cellar/cmake/3.6.2/bin/cmake -P /Volumes/SSD/workspace/osx-kicad-adam-new-head/kicad-mac-builder/build/wxpython-prefix/src/wxpython-stamp/verify-wxpython.cmake
00:03:06.235 CMake Error at wxpython-stamp/verify-wxpython.cmake:22 (message):
00:03:06.235   File not found:
00:03:06.241   /Volumes/SSD/workspace/osx-kicad-adam-new-head/kicad-mac-builder/build/../external/wxPython-src-3.0.2.0.tar.bz2
00:03:06.242 
00:03:06.242 
00:03:06.247 make[2]: *** [wxpython-prefix/src/wxpython-stamp/wxpython-download] Error 1
00:03:06.247 make[1]: *** [CMakeFiles/wxpython.dir/all] Error 2
00:03:06.247 make[1]: *** Waiting for unfinished jobs....
```

I have not had a chance to debug it myself, maybe you can spot the mistake easily.

nickoe, 2018-03-26:
OK, I see. It looks for the folder `external` next to the `build`-dir. But I created the `build`-dir inside `kicad-mac-builder`.

adamwolf, 2018-03-26:
Yeah, I'll make it relative to the source rather than build.

nickoe, 2018-03-26:
In the meantime I made the build dir in the git root folder and called cmake into kicad-mac-builder instead. Now I get this error when configuring kicad itself. It can't find wxpython properly it seems.

```
00:31:48.944 -- Found SWIG: /Users/kicad/homebrew/bin/swig (found suitable version "3.0.10", minimum required is "3.0") 
00:31:49.104 -- Found PythonInterp: /Volumes/SSD/workspace/osx-kicad-adam-new-head/kicad-mac-builder/build/python-dest/bin/python2.7 (found suitable version "2.7.13", minimum required is "2.6") 
00:31:49.104 -- Check for installed Python Interpreter -- found
00:31:49.104 -- Python module install path: /Volumes/SSD/workspace/osx-kicad-adam-new-head/kicad-mac-builder/build/kicad/src/kicad-build/kicad/kicad.app/Contents/Frameworks/python/site-packages
00:31:49.111 -- Found PythonLibs: /Volumes/SSD/workspace/osx-kicad-adam-new-head/kicad-mac-builder/build/python-dest/lib/libpython2.7.dylib (found suitable version "2.7.13", minimum required is "2.6") 
00:31:49.161 -- wxPython version 3.0 found.
00:31:49.202 Traceback (most recent call last):
00:31:49.202   File "<string>", line 1, in <module>
00:31:49.207 ImportError: No module named wx
```
http://ci.kicad-pcb.org/job/osx-kicad-adam-new-head/43/console

adamwolf, 2018-03-26:
I must have another path hardcoded.  While I look for it, could you make a build directory parallel to the ci directory and external and do a build in there?

nickoe, 2018-03-26:
I am not sure what you men with in parallel to the ci dir.  I get the same error when the build dir is in the git root dir. Meaning next to the external dir and also inside the kicad-mac-builder.

adamwolf, 2018-03-26:
I'm wrapping up some work on the unified dmg today, but I need to figure
out why you can't reproduce this build.  Do you have a complete build log?
Let's move this to a new issue.

closed 2018-03-26 by adamwolf