KiCad/kicad-mac-builder #26: Undefined symbols for architecture x86_64:

nickoe, 2018-03-09:
```
17:21:50 [ 32%] Linking CXX shared library ../../../kicad/kicad.app/Contents/Frameworks/libkicad_3dsg.dylib
17:21:51 [ 35%] Built target gal
17:21:51 Undefined symbols for architecture x86_64:
17:21:51   "operator<<(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, wxScopedCharTypeBuffer<char> const&)", referenced from:
17:21:51       S3D::ReadCache(char const*, void*, bool (*)(char const*, void*)) in ifsg_api.cpp.o
17:21:51 ld: symbol(s) not found for architecture x86_64
17:21:51 clang: error: linker command failed with exit code 1 (use -v to see invocation)
17:21:51 make[5]: *** [kicad/kicad.app/Contents/Frameworks/libkicad_3dsg.2.0.0.dylib] Error 1
17:21:51 make[4]: *** [3d-viewer/3d_cache/sg/CMakeFiles/kicad_3dsg.dir/all] Error 2
17:21:51 make[4]: *** Waiting for unfinished jobs....
```

See complete log at http://ci.kicad-pcb.org/job/osx-kicad-adam-new-head/23/console

adamwolf, 2018-03-09:
> osx-kicad-adam-new-head/23/console

Hmm, my builds from last night worked on 10.11 through 10.13.   I'll take a
look!

bstegmaier75, 2018-03-26:
This looks like an issue mixing libstdc++ and libc++. The switch happened around 10.8.
Are all libs built with the same minimum SDK version?

bstegmaier75, 2018-03-26:
Usually, if you don't set anything the OS version is used. So, if you have macOS 10.11 it will use libc++ (e.g. for building wxWidgets).
If you now force KiCad build to something lower than 10.8 (when the switch happened), it will use a libstdc++ and that causes that kind of problems.

However, I am just guessing... I didn't have that problems for years now, because I never did build with such old min versions... but, it was a pain around the switched versions.

adamwolf, 2018-03-26:
Ah, I bet that's it!

Set MACOS_MIN_VERSION please.

Does anyone know if there is a way to make CMake default to the current mac
os version, like I did in shell in build.sh?

`cmake -DMACOS_MIN_VERSION=`sw_vers -productVersion | cut -d. -f1-2``

nickoe, 2018-03-26:
But at other times I see:
```
checking if C compiler (clang -mmacosx-version-min=10.11) works with SDK/version options... yes
checking if C++ compiler (clang++ -mmacosx-version-min=10.11) works with SDK/version options... yes
```
So that is probably a red herring.

adamwolf, 2018-03-26:
> I guess that is not so important, I am just trying with the build.sh
> script to see if it works. But I also noticed below in the logs, maybe this
> is relevant to this issues. This is also seen with the build.sh script.
>
> checking for --with-cxx-main=<compiler>... no
> checking for g++... no
> configure: WARNING:
>
>   By default, distutils will build C++ extension modules with "g++".
>   If this is not intended, then set CXX on the configure command line.

Interesting.

bstegmaier75, 2018-03-26:
I started to always specify clang and clang++ via environment variables (configure.sh) or on cmake command line.

If Xcode clang is the only compiler on the machine gcc and clang should be the same. But, I was never sure if it uses the same libraries if invoked as clang or gcc.

If you happen to have a real gcc on the system, I guess you will definitely run into trouble. At least back then when I tried gcc didn't use libc++.

adamwolf, 2018-03-26:
I mean, if it's something we should always do, let's specify it in this
packaging suite.

Should we just set it in CMakeLists.txt?

bstegmaier75, 2018-03-26:
I wouldn't do that.
It is no problem to use whatever compiler you want (in fact, for OpenMP you would have to specify some specific non-Xcode compiler to use).
You just have to make sure that all dependencies are built with something compatible.
In general this is no problem, the only thing that I know of is C++ stuff with libstdc++/libc++.

nickoe, 2018-03-26:
OK, it seems to build not after I added ``-DMACOS_MIN_VERSION=`sw_vers -productVersion | cut -d. -f1-2` `` as fro the build.sh

adamwolf, 2018-03-26:
I just set up https://github.com/wayneandlayne/kicad-mac-builder/pull/42 so hopefully we won't have this happen again.  Thanks for all the help, and sorry for all the back and forth!  Pretty soon I'll be able to remove the "does not work" from the repo description :)

closed 2018-03-26 by adamwolf