KiCad/kicad-mac-builder #29: wx import issue on kicad build

adamwolf, 2018-03-26:
@nickoe is having problems with kicad finding wx at build time:

"Testing with your fix for the path, it builds wxpython just fine it seems, but now I get this error when configuring kicad itself. It can't find wxpython properly it seems.

00:31:48.944 -- Found SWIG: /Users/kicad/homebrew/bin/swig (found suitable version "3.0.10", minimum required is "3.0") 
00:31:49.104 -- Found PythonInterp: /Volumes/SSD/workspace/osx-kicad-adam-new-head/kicad-mac-builder/build/python-dest/bin/python2.7 (found suitable version "2.7.13", minimum required is "2.6") 
00:31:49.104 -- Check for installed Python Interpreter -- found
00:31:49.104 -- Python module install path: /Volumes/SSD/workspace/osx-kicad-adam-new-head/kicad-mac-builder/build/kicad/src/kicad-build/kicad/kicad.app/Contents/Frameworks/python/site-packages
00:31:49.111 -- Found PythonLibs: /Volumes/SSD/workspace/osx-kicad-adam-new-head/kicad-mac-builder/build/python-dest/lib/libpython2.7.dylib (found suitable version "2.7.13", minimum required is "2.6") 
00:31:49.161 -- wxPython version 3.0 found.
00:31:49.202 Traceback (most recent call last):
00:31:49.202   File "<string>", line 1, in <module>
00:31:49.207 ImportError: No module named wx
http://ci.kicad-pcb.org/job/osx-kicad-adam-new-head/43/console"

adamwolf, 2018-03-26:
@nickoe Can you send me the directory contents of /Volumes/SSD/workspace/osx-kicad-adam-new-head/kicad-mac-builder/build/kicad/src/kicad-build/kicad/kicad.app/Contents/Frameworks/python/site-packages?

nickoe, 2018-03-26:
http://ci.kicad-pcb.org/job/osx-kicad-adam-new-head/ws/build/kicad/src/kicad-build/kicad/kicad.app/Contents/Frameworks/python/site-packages/

adamwolf, 2018-03-26:
Due to not setting MACOS_MIN_VERSION, fixing with https://github.com/wayneandlayne/kicad-mac-builder/pull/42.

closed 2018-03-26 by adamwolf