KiCad/kicad-mac-builder #32: Check help

adamwolf, 2018-03-26:
Are we including help files correctly?  Are there extra or missing files?

craftyjon, 2018-03-26:
Tested on
```
Application: kicad
Version: (5.0.0-rc2-dev-311-g1dd4af297), release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.54.0 SecureTransport zlib/1.2.8
Platform: Mac OS X (Darwin 16.7.0 x86_64), 64 bit, Little endian, wxMac
Build Info:
    wxWidgets: 3.0.4 (UTF-8,STL containers,compatible with 2.8)
    Boost: 1.65.0
    Curl: 7.54.0
    Compiler: Clang 9.0.0 with C++ ABI 1002

Build settings:
    USE_WX_GRAPHICS_CONTEXT=ON
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_ACTION_MENU=OFF
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=OFF
    KICAD_SPICE=OFF
```

Help->KiCad Manual gives 404: `file:///Applications/Kicad/kicad.app/Contents/SharedSupport/help/en/kicad.html/` not found

Help->Getting Started gives 404: `file:///Applications/Kicad/kicad.app/Contents/SharedSupport/help/en/getting_started_in_kicad.html/` not found

Help->Get Involved works properly.

Looks like these files exist in the right place, but the in-app links are appending an extra `/` to the end of the URL which causes it to 404.

craftyjon, 2018-03-26:
Do we want to package all the .pdf and .epub files if we are just going to open the HTML version when user chooses the option from the Help menu?

adamwolf, 2018-03-26:
I think html only is fine.  Thoughts?

craftyjon, 2018-03-26:
I agree, it will make the packages a bit smaller

nickoe, 2018-03-28:
The issue with the tailing slash has been an issue on osx a long time and is not a new issue. FWIW. It is probably a bug in kicad and not the packaging scripts.

craftyjon, 2018-03-28:
Yeah, it is a bug in KiCad for sure.  I'm looking at a fix.

adamwolf, 2018-03-28:
No fair, I've got breakpoints in SearchHelpFileFullPath as we speak.  :)

No worries, I'll work on more packaging stuff.  If you look in VCS history,
you'll see I fixed this sort of issue when 4 came out.  The wheel of
history...

craftyjon, 2018-03-28:
https://git.launchpad.net/kicad/commit/?id=ed99251f0b2683b3496974c105f730ab5a757648 this should do it based on my quick testing

adamwolf, 2018-03-28:
> https://git.launchpad.net/kicad/commit/?id=ed99251f0b2683b3496974c105f730
> ab5a757648 this should do it based on my quick testing

Looks good to me!

craftyjon, 2018-03-28:
OK I just repackaged with this change and help seems to be working fine for `en` locale at least.

adamwolf, 2018-03-28:
Is there a cmake directive to the help subproject that can build just
html?  If not, we can always clean it up in shell :)

adamwolf, 2018-03-30:
> OK, in the latest, English help works because it exists inside the app
> bundle, but other language help doesn't get loaded, with errors like:
> Failed to open URL "file:///Library/Application
> Support/kicad/help/fr/kicad.html" in default browser. (error 2: No such
> file or directory)
> (The file does in fact exist there, if you paste that URL into a browser
> the French help appears)
> This might be another KiCad bug related to spaces or something.

I'm looking into this.

adamwolf, 2018-03-30:
I think the non-bundled help issue might lie inside of wxDoLaunchDefaultBrowser in utils_osx, but I can't debug into my wx, so I am not sure.  I wonder how I'd set that up... any ideas?

craftyjon, 2018-03-30:
This looks promising!  Maybe we can just pull this into our wx branch:
https://github.com/wxWidgets/wxWidgets/commit/5e7db7c0f1b6060c83ba4b9452bb6fc549331926

adamwolf, 2018-03-30:
Do we need to apply it to utils osx or does the x11 one somehow overlay?  (I am pretty new to wx)

craftyjon, 2018-03-30:
Oh oops, wrong one.  This one seems more relevant
https://github.com/wxWidgets/wxWidgets/commit/026659297b20c2c498dc378944b7125ef31834c2

craftyjon, 2018-03-31:
Fixed by e3d9a23 in kicad

adamwolf, 2018-04-06:
Confirmed fixed.  Thanks!

closed 2018-04-06 by adamwolf