KiCad/kicad-mac-builder #36: Think about migration from v4

adamwolf, 2018-03-26:
Is there anything special people need to do to upgrade from v4?  Should they backup their Application Support/kicad directory first?

craftyjon, 2018-03-29:
Really it's too bad that we don't have a better config management system in place (hopefully this will be better for V6) so we should advise that people should back up their config files if they want to maintain preferences, or be prepared to re-set their settings to they way they like it.

adamwolf, 2018-03-29:
Can we make a list of all the config file locations, and then make a script
to make a zip of them?  It might be prudent to do this on other platforms
too.

Beyond config files, I am worried about people overwriting their symbols
and footprints and encountering the new organizations (I think they're
great, I'm not complaining, I just don't want to break users who haven't
changed any system libraries since v4 came out).

Are there other things we need to think about beyond those three?

adamwolf, 2018-06-18:
I poked the list to make sure that we have some general documentation available.

adamwolf, 2018-06-22:
We are discussing this on the KiCad list.  Closing here.

closed 2018-06-22 by adamwolf