KiCad/kicad-mac-builder #39: Check localization

adamwolf, 2018-03-26:
Are we including them correctly? Are there extra or missing files? Do they show up automatically? Do we need to include tables or any configuration files?

craftyjon, 2018-03-29:
The directory structure for the application support package in the DMG seems to be wrong, causing i18n to not work.
It currently is `kicad/ApplicationSupport/<stuff>` but I think it should just be `kicad/<stuff>`

adamwolf, 2018-03-29:
Ah, that's a regression.  Thanks for the note.

adamwolf, 2018-03-30:
> This is now fixed

It seems silly, but as the script changes, I suspect we'll have some
regressions, so for posterity's sake I'll collect a short test procedure
before closing these "check" issues out.

For this, we open up KiCad, go to Preferences -> Language and set it to
something other than English.  We see that it changes the strings, and then
we change it back.

Adam

closed 2018-03-30 by adamwolf