KiCad/kicad-mac-builder #44: Issues on OSX

fellbuendel, 2018-03-27:
Observations from a quick test of build 5.0.0-rc2-dev-315-g8b94606 on OSX:
- Trying to start Pcbnew either from the project selection window or from eeschema quits Kicad (Doesn't show a crash popup, just quits)
- Same for opening the footprint editor
- In Eeschema, using either the component adding tool or the power symbol adding tool immediately quits Kicad. Selecting the tools works, the moment you use them it quits. No crash popup.

Version info:
Application: kicad
Version: (5.0.0-rc2-dev-315-g8b94606), release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.54.0 LibreSSL/2.0.20 zlib/1.2.11 nghttp2/1.24.0
Platform: Mac OS X (Darwin 17.4.0 x86_64), 64 bit, Little endian, wxMac
Build Info:
    wxWidgets: 3.0.4 (UTF-8,STL containers,compatible with 2.8)
    Boost: 1.61.0
    Curl: 7.43.0
    Compiler: Clang 7.3.0 with C++ ABI 1002

Build settings:
    USE_WX_GRAPHICS_CONTEXT=ON
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_ACTION_MENU=OFF
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=OFF
    KICAD_SPICE=OFF

nickoe, 2018-03-27:
I still see issues with wxpython when building, maybe this is related to the crashes you see.

```
00:03:25.343 -- wxPython version 3.0 found.
00:03:25.438 Traceback (most recent call last):
00:03:25.438   File "<string>", line 1, in <module>
00:03:25.438 ImportError: No module named wx
```

adamwolf, 2018-03-27:
Nick, can I make another set of credentials to upload files so we can do a
differential analysis?

Who do I talk to about that again?

Adam

adamwolf, 2018-03-29:
@craftyjon Are you able to verify some things for me?  When I run my KiCad.app through a debugger, and open up Pcbnew from the project selector, I'm getting a crash in wxString GetOSXKicadUserDataDir(), at     wxFileName udir( wxStandardPaths::Get().GetUserDataDir(), wxEmptyString );.  Is that real, or is my debugger messed up?

craftyjon, 2018-03-29:
I don't see that.  Using lldb 900.0.64 I'm able to debug without that crash.

adamwolf, 2018-03-29:
Bonkers.  I added some debug code so I could see what wxStandardPaths::Get() returned, and then what wxStandardPaths::Get().GetUserDataDir() returns, and it worked.  I uncommented my debug code, and now I can open pcbnew.

Can you open pcbnew, @craftyjon?

craftyjon, 2018-03-29:
Yes, I can debug pcbnew under lldb no problem, either by targeting kicad.app and hitting the pcbnew button or by directly targeting the pcbnew binary inside the app bundle.

craftyjon, 2018-03-29:
OK, I installed the latest nightly (8b94606) and now I see this issue (pcbnew crashing/closing)!  I wonder why this doesn't happen when I build locally.

craftyjon, 2018-03-29:
@adamwolf does this line up with what you saw?

Jenkins build:
```
Jons-MacBook-Pro:debug jevans$ lldb /Applications/Kicad/kicad.app/Contents/Applications/pcbnew.app/Contents/MacOS/pcbnew
(lldb) target create "/Applications/Kicad/kicad.app/Contents/Applications/pcbnew.app/Contents/MacOS/pcbnew"
Current executable set to '/Applications/Kicad/kicad.app/Contents/Applications/pcbnew.app/Contents/MacOS/pcbnew' (x86_64).
(lldb) r
Process 43984 launched: '/Applications/Kicad/kicad.app/Contents/Applications/pcbnew.app/Contents/MacOS/pcbnew' (x86_64)
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
ImportError: No module named site
Process 43984 exited with status = 1 (0x00000001)
```
Home-built:
```
Jons-MacBook-Pro:debug jevans$ lldb ~/src/bin/kicad.app/Contents/Applications/pcbnew.app/Contents/MacOS/pcbnew
(lldb) target create "/Users/jevans/src/bin/kicad.app/Contents/Applications/pcbnew.app/Contents/MacOS/pcbnew"
Current executable set to '/Users/jevans/src/bin/kicad.app/Contents/Applications/pcbnew.app/Contents/MacOS/pcbnew' (x86_64).
(lldb) r
Process 43991 launched: '/Users/jevans/src/bin/kicad.app/Contents/Applications/pcbnew.app/Contents/MacOS/pcbnew' (x86_64)
09:56:28 AM: Debug: Adding locale lookup path: /Library/Application Support/kicad/share/internat
Process 43991 exited with status = 0 (0x00000000)
```

adamwolf, 2018-03-29:
I'm not sure what's happening different on Nick's Jenkins build.  I know
he's on 10.11, but I've got builds I've made on 10.11 as well that are
working for me.  This is the main thing I'm digging into.

nickoe, 2018-04-03:
@adamwolf I don't intend to touch the setup currently, because it is currently serving the old nightlies.

I may try to setup a new/updated toolchain (homebrew) and run your script with that to see if it changes anything.

adamwolf, 2018-04-03:
Please don't break the current nightlies :)

Adam

nickoe, 2018-04-10:
@craftyjon Any update here? Are you using the builds from http://downloads.kicad-pcb.org/osx/testing/ when you say jenkins?

craftyjon, 2018-04-10:
Sorry, I've been away from KiCad stuff since last week, just getting back to things.  I installed `kicad-unified-20180410-035730-94dbcc7.dmg` on my 10.12 machine and I see the exact same behavior as in my previous comment.  PcbNew exits immediately with no error messages and return code of 1 when I try to launch it.  KiCad launcher works properly.

adamwolf, 2018-04-20:
This is reproducible by me when building on 10.11 as well.

lldb says

Process 23881 launched: '/Users/wolf/KiCad/kicad.app/Contents/MacOS/kicad' (x86_64)
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
ImportError: No module named site
Process 23881 exited with status = 1 (0x00000001)

I'll dig into this, folks. This is one of the last blockers!

adamwolf, 2018-04-23:
Many of these are fixed in #72.

[image]

If you are still interested, please let me know how they are after tonight's nightly gets built.

adamwolf, 2018-04-23:
I merged some Python changes into master.  It changes these bugs a bit, and closes some of them, but opens other ones.  (Scripting console works, but you have to open pcbnew.app standalone, for example.) Working on it.

adamwolf, 2018-04-26:
Could someone else test one of the nightlies from http://downloads.kicad-pcb.org/osx/testing/ from April 26?

All of these issues are fixed for me there.

craftyjon, 2018-04-26:
Yes with the 4/26 nightly I can now use pcbnew!  Nice work.

closed 2018-04-26 by adamwolf