KiCad/kicad-mac-builder #46: Make sure wx is importable by KiCad's Python

adamwolf, 2018-03-28:
During the CMake step, some folks are getting something like

```
-- Check for installed Python Interpreter -- found
-- Python module install path: /Users/wolf/wnl/kicad/clion/kicad/cmake-build-debug/kicad/kicad.app/Contents/Frameworks/python/site-packages
-- wxPython version 3.0 found.
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: No module named wx
```

nickoe, 2018-03-31:
Maybe we need something like we have in the old builder:

"-DPYTHON_SITE_PACKAGE_PATH=" + os.path.realpath("wx/wx-bin/lib/python2.7/site-packages"),

adamwolf, 2018-03-31:
I think we do, in CMakeLists.txt.

set( KICAD_CMAKE_ARGS ...   -DPYTHON_SITE_PACKAGE_PATH=${
wxwidgets_INSTALL_DIR}/lib/python2.7/site-packages ... )

I suspect the issue here is that we have installed wx into the python that
we're going to compile KiCad against and bundle with it, but not into the
Python that cmake is invoking during configure time.  I am not sure this is
actually a problem, beyond making noise that says 'error'.  I am going to
see if I can aim CMake at that python, and see if the error goes away
though.

adamwolf, 2018-04-06:
@nickoe this should be fixed in master, please confirm after the nightlies and we can close.

nickoe, 2018-04-19:
Well, it is building now. So I guess this can be closed.

nickoe, 2018-04-22:
Please close this.

adamwolf, 2018-04-22:
Really?  The message may be gone during build, but wx isn't importable from
KiCad's python yet.

nickoe, 2018-04-22:
Originally when I mentioned this issue, somewhere else, you created this issue to track it. I thought you wanted to track the rest of the issue in https://github.com/wayneandlayne/kicad-mac-builder/issues/9. Either is fine for me though...

adamwolf, 2018-04-23:
That's right.

PR #68 (2018-04-06): Use the right Python to install wxpython
@nickoe this looks like it fixes #46. I wonder if it makes your binaries usable by everyone...
closed 2018-04-23 by adamwolf