KiCad/kicad-mac-builder #48: wx configure runs every time you try to build

craftyjon, 2018-03-29:
Whenever I run `build.sh` I have to wait a long time for wxwidgets `configure` to run, even if the wxwidgets build is up to date.  Not sure how easy this would be to fix, but if we could fix it, the script runtime would go down a lot when re-testing minor changes.

adamwolf, 2018-03-29:
Seems related to https://gitlab.kitware.com/cmake/cmake/issues/16419, but I *already* have `UPDATE_DISCONNECTED 1`.  I wonder if it'll help if I set it to a tag, rather than a branch.

adamwolf, 2018-03-29:
For development, you could set UPDATE_COMMAND to "".

adamwolf, 2018-03-29:
This is slightly confusing to me.
```
[  0%] Skipping update step for 'wxwidgets'
cd /Users/wolf/wnl/kicad/kicad-mac-builder/build/wxwidgets/src/wxwidgets && /usr/local/Cellar/cmake/3.10.2/bin/cmake -E echo_append
[ 38%] Performing configure step for 'wxwidgets'
cd /Users/wolf/wnl/kicad/kicad-mac-builder/build/wxwidgets/src/wxwidgets && CPPFLAGS=-D__ASSERT_MACROS_DEFINE_VERSIONS_WITHOUT_UNDERSCORES=1 MAC_OS_X_VERSION_MIN_REQUIRED=10.13 ./configure --prefix=/Users/wolf/wnl/kicad/kicad-mac-builder/build/wxwidgets-dest --with-opengl --enable-aui --enable-utf8 --enable-html --enable-stl --disable-mediactrl --with-libjpeg=builtin --with-libpng=builtin --with-regex=builtin --with-libtiff=builtin --with-zlib=builtin --with-expat=builtin --without-liblzma --with-macosx-version-min=10.13 CC=clang CXX=clang++
```
If it's skipping the update step, why would it ever do the configure step?  *sigh*

closed 2018-06-29 by adamwolf