KiCad/kicad-mac-builder #53: 'demos/' in KiCad/ in dmg?

adamwolf, 2018-03-30:
What are these, and where do they actually belong?

craftyjon, 2018-03-31:
These are the example projects.  Maybe they should be part of kicad-extras instead of kicad?

adamwolf, 2018-04-03:
I'm fine putting them in the extras and unified dmgs only.  Can we put them in the Application Support/kicad directory?

craftyjon, 2018-04-03:
If we put them in Application Support, they will require admin access to modify.  Is that what we want?  I could see people opening up the demos and then being confused when KiCad complains about not being able to save them.

adamwolf, 2018-04-03:
I'm not sure.  I can see people not wanting them to be overwritten, I guess, so maybe it's a feature? :)

adamwolf, 2018-05-04:
OK, sounds like we should pull these out of Applications, and put them top level in both kicad-unified and kicad-extras, without an arrow pointing to where they should go.

adamwolf, 2018-06-17:
I actually think I'm not going to put them in extras.  They're really small, 10M or something.  I'm going to put them in nightly and in unified.

I have initial work for this merged in.  I need to modify the DMG to make everything pretty, but I have already captured that in #143

closed 2018-06-17 by adamwolf