KiCad/kicad-mac-builder #119: Kicad build machine can't download packages3d

adamwolf, 2018-05-14:
*sigh*

http://ci.kicad-pcb.org/job/osx-kicad-adam-new-head/132/console

```13:06:49 [ 18%] Performing download step (git clone) for 'packages3d'
13:06:49 cd /Volumes/SSD/workspace/osx-kicad-adam-new-head/build/packages3d/src && /Users/kicad/homebrew/Cellar/cmake/3.6.2/bin/cmake -P /Volumes/SSD/workspace/osx-kicad-adam-new-head/build/packages3d/tmp/packages3d-gitclone.cmake
13:06:49 fatal: could not create work tree dir '': No such file or directory
13:06:49 fatal: could not create work tree dir '': No such file or directory
13:06:49 fatal: could not create work tree dir '': No such file or directory
13:06:49 -- Had to git clone more than once:
13:06:49           3 times.
13:06:49 CMake Error at /Volumes/SSD/workspace/osx-kicad-adam-new-head/build/packages3d/tmp/packages3d-gitclone.cmake:56 (message):
13:06:49   Failed to clone repository: 'https://github.com/KiCad/kicad-packages3D.git'
13:06:49 
13:06:49 
13:06:49 make[3]: *** [packages3d/src/packages3d-stamp/packages3d-download] Error 1
13:06:49 make[2]: *** [CMakeFiles/packages3d.dir/all] Error 2
```

Figure out what this error is, and see if I can reproduce it.

adamwolf, 2018-05-15:
I got successful builds on my build machine with #120.

adamwolf, 2018-05-15:
@nickoe I merged #120 into master, so if you are able to trigger a build sometime this morning I could check on it and make any needed changes tonight.

nickoe, 2018-05-15:
I just triggered a build.

nickoe, 2018-05-15:
The build succeeded. You can probably close this issue. :)

adamwolf, 2018-05-15:
WOOO!

closed 2018-05-15 by adamwolf