KiCad/kicad-mac-builder #123: PCBnew doesn't work on nickoe builds

adamwolf, 2018-05-18:


adamwolf, 2018-05-18:
Working on this today.

adamwolf, 2018-05-18:
This is an issue where the rpath for Python things is empty.  We explicitly add the rpath, and we verify it's set.  I need to read the build logs for the nickoe stuff to make sure that is happening.

adamwolf, 2018-05-18:
I am seeing a few issues.  The Python framework's Current symlink is turning into a copy during fixup_bundle, and the rpath is gone on the binaries after fixup_bundle.  I am almost certain these were not the case before I backported the BundleUtilities from 3.10.x to 3.6.  I am reverting my build system to 3.10.2 and seeing if this is a regression or something more serious.

adamwolf, 2018-05-19:
OK, fixed all the issues I found, like fixup_bundle clobbering some rpaths and destroying symlinks, and there's *some* reference to python-dest at this point.  I need to find it, and make it relative.

adamwolf, 2018-05-19:
I think I fixed it in #127 !  @nickoe, please remove the build/src/kicad directory (or the whole build/ directory) before builds tonight, as I modified one of the kicad patches in this.

nickoe, 2018-05-19:
Ok. Done.

http://ci.kicad-pcb.org/job/osx-kicad-adam-new-head/145/console

adamwolf, 2018-05-20:
It built.  I'll download it, check that it works, and hopefully close out
some issues!

adamwolf, 2018-05-20:
Fixed!

closed 2018-05-20 by adamwolf