KiCad/kicad-mac-builder #126: Enabling ngspice makes the build fail

nickoe, 2018-05-19:
Since https://github.com/wayneandlayne/kicad-mac-builder/commit/0c447cff34a708d2077d784728e3a59227c28810

I see this build failure:
```
00:04:25.803 error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: no LC_RPATH load command with path: /Volumes/SSD/workspace/osx-kicad-adam-new-head/build/wxwidgets-dest/lib found in: /Volumes/SSD/workspace/osx-kicad-adam-new-head/build/kicad-dest/kicad.app/Contents/MacOS/kicad (for architecture x86_64), required for specified option "-delete_rpath /Volumes/SSD/workspace/osx-kicad-adam-new-head/build/wxwidgets-dest/lib"
00:04:35.582 -- fixup_bundle
```

I am not sure how that error messages is related to the enabling of ngpsice.

adamwolf, 2018-05-19:
> Since 0c447cf

That isn't the error--the actual error is in the bundle step.  I was
looking at that on Friday (
https://github.com/wayneandlayne/kicad-mac-builder/issues/96), when I
realized that pcbnew on your builds wasn't working.  I got that fixed, so
next is to get that through my build cluster and merged to master.  Once
that's done, if it's close to the end of the day, I'll just disable ngspice
again so I can see the results from your builds tomorrow--but I hope to fix
both today.  Thanks for keeping an eye on it--we are SO CLOSE!

adamwolf, 2018-05-25:
Fixed.

closed 2018-05-25 by adamwolf