KiCad/kicad-mac-builder #135: Build broken again

nickoe, 2018-06-04:
It looks like the latest commit https://github.com/wayneandlayne/kicad-mac-builder/commit/36607c47e2697aee4e89ffe0cddfffe96779c748 broke the build, see
http://ci.kicad-pcb.org/job/osx-kicad-adam-new-head/177/

With stuff like:
```
00:00:35.837 Parse error.  Function missing ending ")".  End of file reached.
00:00:35.837 CMake Error at CMakeLists.txt:68 (include):
00:00:35.837   include could not find load file:
00:00:35.837 
00:00:35.837     footprints.cmake
00:00:35.837 
00:00:35.837 
00:00:36.063 CMake Error at /Users/kicad/homebrew/Cellar/cmake/3.6.2/share/cmake/Modules/ExternalProject.cmake:2140 (get_property):
```

nickoe, 2018-06-04:
It is probably caused by removal of this ending parenthesis

https://github.com/wayneandlayne/kicad-mac-builder/commit/36607c47e2697aee4e89ffe0cddfffe96779c748#r29234082

adamwolf, 2018-06-04:
> 36607c4#r29234082

Sorry, will fix shortly.

closed 2018-06-06 by adamwolf