KiCad/kicad-mac-builder #137: STEP export uses wrong path

sethhillbrand, 2018-06-09:
Exporting a STEP file from pcbnew tries to execute the command:
/Applications/KiCad/kicad.app/Contents/Applications/pcbnew.app/Contents/MacOS/kicad2step

But the application is actually
/Applications/KiCad/kicad.app/Contents/MacOS/kicad2step

adamwolf, 2018-06-09:
Thanks!  Can you write up instructions on how to export STEP, or link me to
some simple ones?

Adam

sethhillbrand, 2018-06-09:
Here's a simple board file. 

Open pcbnew (stand alone)
Open this file
Go to File->Export->STEP
[led-pannel2.kicad_pcb.zip](https://github.com/wayneandlayne/kicad-mac-builder/files/2087192/led-pannel2.kicad_pcb.zip)

adamwolf, 2018-06-12:
I think this is an issue in KiCad, actually.

in dialog_export_step.cpp,

it says

```
    wxFileName appK2S( wxStandardPaths::Get().GetExecutablePath() );
    appK2S.SetName( "kicad2step" );
```

Really, we need to get the path of the kicad main app, if we are on macos.  I will try to make a KiCad patch tomorrow for this.

sethhillbrand, 2018-06-12:
OK.  Thanks for digging into this.

PR #138 (2018-06-13): Added kicad patch to fix step when launched from standalone pcbnew.
Fixes #137.
closed 2018-06-12 by sethhillbrand