KiCad/kicad-mac-builder #156: build.sh package-kicad-nightly downloads docs. Is this the intended behaivour?

nickoe, 2018-06-25:
Is it intentional that the build configuration:

`build.sh package-kicad-nightly creates a DMG of "mostly just KiCad" in build/dmg.`

Also downloads the docs?

adamwolf, 2018-06-25:
Yes.  It needs to download the docs or else the help menus don't work.  It
is really just packages3d that is missing.

adamwolf, 2018-06-25:
I'll clarifying in a DMG cleanup PR that will be up today.

closed 2018-06-25 by adamwolf

nickoe, 2018-06-25:
Ok, I was expecting it to only build kicad itself, and then stuff it in a dmg for redistribution. Intended to test pactches quickly. But ok.

adamwolf, 2018-06-25:
Yeah, that's what I wanted it to be, but then a lot of things in KiCad
don't work, and as thin as the macOS team is, I'd rather not have to
support folks using nightlies and running into "known broken" things.

I hope that kicad-mac-builder makes it so each that lots and lots of mac
devs join and help :)