KiCad/kicad-mac-builder #157: Build error on ci.kicad-pcb.org

nickoe, 2018-06-25:
I see this build error from http://ci.kicad-pcb.org/job/osx-kicad-adam-new-head/221/consoleFull

I tried to clean the worksapce and try again to see if it was something transient in the build scripts. It sitll failed with various errors:

I don't understand this one:
`00:00:24.691 fatal: could not create work tree dir '': No such file or directory`

Please look in the log for details. Is there a switch to not make the build targets parallel for debugging?

adamwolf, 2018-06-25:
There is a script already meant for ci (ci/src/build.sh) that should clear out the workspace and rebuild.  It would be nice if that script overrode the number of jobs for the second build so that it only built with -j1.

I will take a look at both of these things.  I am working on the stable switch to build.sh right now, but I will look at the output while this build is going on.

nickoe, 2018-06-25:
It looks like it is ngspice that fails.
```
00:00:24.969 fatal: could not create work tree dir '': No such file or directory
00:00:24.986 fatal: could not create work tree dir '': No such file or directory
00:00:25.027 fatal: could not create work tree dir '': No such file or directory
00:00:25.032 -- Had to git clone more than once:
00:00:25.032           3 times.
00:00:25.038 CMake Error at /Volumes/SSD/workspace/osx-kicad-adam-new-head/build/packages3d/tmp/packages3d-gitclone.cmake:56 (message):
00:00:25.038   Failed to clone repository: 'https://github.com/KiCad/kicad-packages3D.git'
00:00:25.038 
00:00:25.038 
00:00:25.038 make[3]: *** [packages3d/src/packages3d-stamp/packages3d-download] Error 1
00:00:25.042 make[2]: *** [CMakeFiles/packages3d.dir/all] Error 2
00:00:25.043 make[2]: *** Waiting for unfinished jobs....
```

adamwolf, 2018-06-25:
That looks to me like it can't download packages3d.  I am almost finished
with the single-core-rebuilder.

nickoe, 2018-06-25:
But it does not explain the could not create work tree dir thing.

Also, I assume it caches the kicad-packages3D such that we don't need to make a clean clone everytime.

adamwolf, 2018-06-25:
I thought you said you cleaned the workspace.  If so, it won't have a cache.

adamwolf, 2018-06-25:
Anyway, my four builds today all built fine, both with clean workspace and without, on 10.11 and 10.12, and my dev laptop is on 10.13 and has been building fine too.

I will look into it, but even if it was a transient network error that cleaned itself up, it looks like clearing the workspace in Jenkins can cause problems like this!  https://issues.jenkins-ci.org/browse/JENKINS-16033

adamwolf, 2018-06-25:
I know what it is.  The git on that build machine doesn't support whatever
cmake is doing for GIT_PROGRESS.

I turned that on this weekend.

I will turn it off tonight.

After V5 comes out, let's look at upgrading git and cmake on that machine.

adamwolf, 2018-06-26:
Pushed a fix to master.  Let's see it works.

nickoe, 2018-06-26:
Ok, I guess it was the GIT_PROGRESS thing, it works now.

closed 2018-06-26 by nickoe

adamwolf, 2018-06-26:
Awesome!