KiCad/kicad-mac-builder #161: Wxwidgets is always rebuilt for target kicad

nickoe, 2018-06-26:
Hi

It looks lige wxwidgets is rebuilt every time kicad is built. I was wondering about this because I saw this triend on the build times on jenkins.

[image]

The good last good build before the failures shortly before the build time increase is #176 with 3afb7bc7e17415f2d30a49a7fa810e3cb1f62eab and the first good build with long build times are #180 with e652a42a896355e7aee9233dbf94791edd6c93c2.

I don'ẗ think this is intendended, and this is sort of against using this for easy contributions from osx people.

I originally started to look at this because I wanted to test a patch, where it seems like it is not applied when I add a `.patch` file in `kicad-mac-builder/patches/kicad/`

nickoe, 2018-06-26:
It looks like the main change is the removal of that `UPDATE_DISCONNECTED` stuff in https://github.com/wayneandlayne/kicad-mac-builder/commit/36607c47e2697aee4e89ffe0cddfffe96779c748#diff-6dca1159af1638174248c8241c9f2928L32

I am not sure I understand why we need to remove this. If you want to rebuild wxwidgets on a CI you could just make sure the CI script will call the `wxwidgets-update `target.

adamwolf, 2018-06-26:
> stuff in 36607c4#diff-6dca1159af1638174248c8241c9f2928L32

It doesn't matter to me.

I removed the update_disconnected stuff when I was debugging something
else.  I wanted to make sure that clean builds work always, and I wanted to
know if wxwidgets pulled the tarball for example.  However, I maintain a
build machine that does it from scratch inside a brand new VM every single
build, so if this makes it easier for you, I'm fine putting it back in.

closed 2018-06-29 by adamwolf