KiCad/kicad-mac-builder #162: Not enough info about patches applied to kicad

nickoe, 2018-06-26:
Related to https://github.com/wayneandlayne/kicad-mac-builder/issues/161 I mentioned that it does not seem like patches are applied.

I determine this because I tried to apply a patch and I could not see any messages suggesting that the patches have been applied to kicad. 

An example on out put I would like to see in the log:
```
git am ../../../../kicad-mac-builder/patches/kicad/random.patch
Applying: OpenGL: Adjust color order and size of buffer elements
```

This is what I did manually, and I don't see that `Applying:` in the console anywhere. On top of that it looks like ther are no messages when this is run successfully.

https://github.com/wayneandlayne/kicad-mac-builder/blob/ff3cbe0b158a01842eadbca4bda8ecab7ef78073/kicad-mac-builder/bin/git-multipatch.sh

In http://ci.kicad-pcb.org/job/osx-kicad-adam-new-head/224/consoleFull I would like to have seen some message that `random.patch` got applied.

nickoe, 2018-06-26:
Ohh, on a second readthrough I see that it prints the file... but I don't see that in the log.
https://github.com/wayneandlayne/kicad-mac-builder/blob/ff3cbe0b158a01842eadbca4bda8ecab7ef78073/kicad-mac-builder/bin/git-multipatch.sh#L9

nickoe, 2018-06-26:
I just pulled to 386d45c45c6ad5f7476248b5eb7ccd9a26df9420 and it still does not seem to work.

When I go to my_build_dir/kicad/src/kicad and do a git log, I don't see any of the patches appled. It does not even pull from upstream.

adamwolf, 2018-06-26:
> I just pulled to 386d45c and it still does not seem to work.

It is supposed to print the file and the git am output.  If you aren't
seeing it, git-multipatch.sh is probably not being called.

I will investigate this.  This is what I see on both my dev laptop and
2 build machines as the top 3 results to `git log` in
build/kicad/src/kicad:

```
commit b0079837c418746abbe0bd9773b8e362e6c50c73 (HEAD -> master)
Author: Adam Wolf <adamwolf@feelslikeburning.com>
Date:   Sat May 19 21:48:25 2018 -0500

    Fixup ngspice library when bundling on macOS.

commit 7474ee117bf935b4482c227214686197465d7a65
Author: Adam Wolf <adamwolf@feelslikeburning.com>
Date:   Thu May 10 07:36:33 2018 -0500

    Backported BundleUtilities and GetPrerequisites from Cmake 3.11.1
to let macOS packaging work at least back to 3.6.2.

    This fixes an issue during bundling where otool exits with an
error complaining about Python linked via an rpath.

commit 0b984063914fb7c31d908cdf05f09d57b8f488aa
Author: Adam Wolf <adamwolf@feelslikeburning.com>
Date:   Sat May 19 15:06:40 2018 -0500

    Setup PYTHON_FRAMEWORK embedded in MacOS.
```

adamwolf, 2018-06-27:
This is going to be a fun puzzle.  What could be wrong with your setup?

When I look at the output of my builds...

```
$ grep 'patch step' consoleTextmine.txt
==> default: [  4%] No patch step for 'docs'
==> default: [  5%] No patch step for 'packages3d'
==> default: [ 17%] No patch step for 'ngspice'
==> default: [ 20%] No patch step for 'wxwidgets'
==> default: [ 28%] Performing patch step for 'python'
==> default: [ 37%] No patch step for 'six'
==> default: [ 44%] No patch step for 'translations'
==> default: [ 52%] No patch step for 'templates'
==> default: [ 58%] No patch step for 'symbols'
==> default: [ 66%] No patch step for 'footprints'
==> default: [ 75%] No patch step for 'wxpython'
==> default: [ 83%] Performing patch step for 'kicad'
==> default: [ 96%] No patch step for 'package-kicad-unified'
==> default: [ 95%] No patch step for 'package-kicad-nightly'
==> default: [ 66%] No patch step for 'package-extras'
```

```
$ grep 'patch step' consoleTextnick.txt

... nothing.

adamwolf, 2018-06-27:
tomorrow, I'm going to get my 10.11 clean VM build machine to upload to testing/.  I will definitely solve this, but I don't want to hold up the release.

nickoe, 2018-06-27:
What cmake version is it that you are running? I will check min and potentially try to update.

nickoe, 2018-06-27:
@adamwolf Hmm, does the build.sh update the kicad source automatically if you are not doing a clean build? I just tried it manually from an old checkout of kicad-mac-builder, whre I did pull the latest changes, but it seems like it didn't update the kicad source. Was still back to april. I am doing a completely clean build there now to test.

nickoe, 2018-06-27:
I mean in my ci I am just doing 

`./ci/src/build.sh package-kicad-unified package-kicad-nightly package-extras`

Which seems to update the kicad source just fine, even before the removal of the UPDATE_DISCONNECTED stuff.

adamwolf, 2018-06-28:
UPDATE_COMMAND "" in kicad.cmake prevents it from updating.

I'm cleaning this up today, and I should be able to make wxwidgets not recompile all the time as well.

closed 2018-06-29 by adamwolf