KiCad/kicad-mac-builder #171: See if this works for developers

adamwolf, 2018-07-06:
Hi folks!

This is getting ready to show to the world.  I run it on my development machine, and on some VMs via Jenkins, but I'd like to see how it works on a better cross-section of systems.

You may be able to help out!

You will need to be running on a Mac, have homebrew installed, have ~20G disk space free, have unmetered Internet, and have a few minutes to start a test (which will take a few minutes to a few hours to complete). It won't install KiCad on your Mac, and it's easy to remove from your computer when you are done.

If you are interested, check out this git respository, take look at the README to setup the dependencies via brew, and use the Terminal to run build.py.

`./build.py --target build-kicad-unified
`

When it finishes and prints "Build complete.", check in `build/dmg` for a DMG file that starts with kicad-unified-.  If that file exists, awesome!  Post here, and include the output from running `sw_vers` in the terminal.  If it doesn't work, troubleshooting it here actually isn't going to be super helpful, so please get in touch with me and we can work through it.

You may want to set the notifications on this issue to "Not Watching", so you are only notified if someone @s you, rather than any time anyone posts here.

jamesneal, 2018-07-06:
Did you mean "./build.py --target package-kicad-unified" above?

adamwolf, 2018-07-06:
Ahhh! yes.

jamesneal, 2018-07-06:
$ sw_vers 
ProductName:	Mac OS X
ProductVersion:	10.13.6
BuildVersion:	17G62a

$ time ./build.py --target package-kicad-unified
[.....]
Build complete.  If you built a DMG, it should be located in /Users/neal/git/kicad-mac-builder/build/build/dmg

real	43m12.301s
user	192m9.208s
sys	19m10.820s

$ ls -l build/dmg/
total 2228232
-rw-r--r--@ 1 neal  staff  1134987368 Jul  5 20:14 kicad-unified-20180705-201148-c5139a1b8.dmg

tednaleid, 2018-07-06:
Other than a few homebrew hiccups (directory permissions that were my fault), it went smoothly on my mac-mini.

I did notice that the path on the last message seems to be incorrect and has an extra `build` in the path:

```
Build complete.  If you built a DMG, it should be located in /Users/tnaleid/Dropbox/workspace/misc/kicad-mac-builder/build/build/dmg
```
(it's just under `kicad-mac-builder/build/dmg`)

results:
```
ls -l build/dmg
total 2228232
-rw-r--r--@ 1 tnaleid  staff  1134528853 Jul  5 23:50 kicad-unified-20180705-234642-8e3eb7dc5.dmg
```

```
sw_vers
ProductName:	Mac OS X
ProductVersion:	10.13.5
BuildVersion:	17F77
```

adamwolf, 2018-07-06:
Thanks folks!  I made new issues to capture the snags.

bengartner, 2018-07-06:
```
09:51 $ ls build/dmg/
kicad-unified-20180705-234901-b29996a6e.dmg
✔ ~/kicad-mac-builder [master|✔] 
09:52 $ sw_vers
ProductName:	Mac OS X
ProductVersion:	10.13.5
BuildVersion:	17F77
✔ ~/kicad-mac-builder [master|✔] 

```
Worked fine for me. It complained about CMake < 3.6.2, but that was before I actually installed any of the dependencies because I was reading this issue, not the readme.

closed 2018-07-20 by adamwolf