KiCad/kicad-mac-builder #193: Unclear how to specify compiler version

bson, 2018-07-29:
How do I specify the compiler to use?  By default the builder uses the Apple Xcode clang/llvm toolchain, which fails.  The KiCAD code seems to depend on specific compiler builtins, more specifically that constant atan() expressions can be evaluated at compile time. 

OSX 10.13.3
Apple LLVM version 9.1.0 (clang-902.0.39.2)

I do have gcc 8 installed...

Otherwise it seems to work very well!

**[ 98%] Building CXX object qa/geometry/CMakeFiles/qa_geometry.dir/test_fillet.cpp.o
In file included from /Users/bson/src/kicad-mac-builder/build/kicad/src/kicad/qa/geometry/test_fillet.cpp:32:
/Users/bson/src/kicad-mac-builder/build/kicad/src/kicad/qa/geometry/geom_test_utils.h:33:18: error: constexpr variable 'PI' must be initialized by a constant expression
constexpr double PI = atan(1.0) * 4.0;
                 ^    ~~~~~~~~~~~~~~~
/Users/bson/src/kicad-mac-builder/build/kicad/src/kicad/qa/geometry/geom_test_utils.h:33:23: note: non-constexpr function 'atan' cannot be used in a constant expression
constexpr double PI = atan(1.0) * 4.0;
                      ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk/usr/include/math.h:331:15: note: declared here
extern double atan(double);
              ^
In file included from /Users/bson/src/kicad-mac-builder/build/kicad/src/kicad/qa/geometry/test_fillet.cpp:32:
/Users/bson/src/kicad-mac-builder/build/kicad/src/kicad/qa/geometry/geom_test_utils.h:34:18: error: constexpr variable 'PI_2' must be initialized by a constant expression
constexpr double PI_2 = atan(1.0) * 2.0;
                 ^      ~~~~~~~~~~~~~~~
/Users/bson/src/kicad-mac-builder/build/kicad/src/kicad/qa/geometry/geom_test_utils.h:34:25: note: non-constexpr function 'atan' cannot be used in a constant expression
constexpr double PI_2 = atan(1.0) * 2.0;
                        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk/usr/include/math.h:331:15: note: declared here
extern double atan(double);
              ^
2 errors generated.
make[6]: *** [qa/geometry/CMakeFiles/qa_geometry.dir/test_fillet.cpp.o] Error 1
make[5]: *** [qa/geometry/CMakeFiles/qa_geometry.dir/all] Error 2
make[4]: *** [all] Error 2
make[3]: *** [kicad/src/kicad-stamp/kicad-build] Error 2
make[2]: *** [CMakeFiles/kicad.dir/all] Error 2
make[1]: *** [CMakeFiles/kicad.dir/rule] Error 2
make: *** [kicad] Error 2
Error while running make after rebuilding with a single job. Please report this issue if you cannot fix it after reading the README.
Traceback (most recent call last):
  File "./build.py", line 212, in <module>
    main()
  File "./build.py", line 208, in main
    build(parsed_args)
  File "./build.py", line 184, in build
    subprocess.check_call(make_command, env=dict(os.environ, PATH=new_path))
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 540, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '-j1', 'kicad']' returned non-zero exit status 2**

adamwolf, 2018-07-29:
> last): File "./build.py", line 212, in main() File "./build.py", line 208,

This is an issue in KiCad. There is no expectation that users use anything
but clang/llvm, and if there was, it would either be done automatically or
documented :)

Those geometry tests are a day or two old, and last night, the automatic
builder didn't run because it was too low on disk space.  I've increased
the disk space, and emailed the dev list re the issue.

Thanks!

bson, 2018-07-30:
Ahh, ok, that makes sense.  Thanks!

closed 2018-07-30 by bson