KiCad/kicad-mac-builder #195: Build fails on doc install

bson, 2018-07-30:
With the fixes to the test cases (I saw the commit) I now get to where it tries to construct the app bundle. Again, running `build.py --target kicad`...  Looks like some help files might be missing perhaps?

> [ 90%] Installing docs into KiCad.app
> mkdir -p /Users/bson/src/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/SharedSupport/help/
> cp -r /Users/bson/src/kicad-mac-builder/build/docs/share/doc/kicad/help/en /Users/bson/src/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/SharedSupport/help/
> cp: /Users/bson/src/kicad-mac-builder/build/docs/share/doc/kicad/help/en: No such file or directory
> make[3]: *** [kicad/src/kicad-stamp/kicad-install-docs-to-app] Error 1
> make[2]: *** [CMakeFiles/kicad.dir/all] Error 2
> make[1]: *** [CMakeFiles/kicad.dir/rule] Error 2
> make: *** [kicad] Error 2
> Error while running make after rebuilding with a single job. Please report this issue if you cannot fix it after reading the README.
> Traceback (most recent call last):
>   File "./build.py", line 212, in <module>
>     main()
>   File "./build.py", line 208, in main
>     build(parsed_args)
>   File "./build.py", line 184, in build
>     subprocess.check_call(make_command, env=dict(os.environ, PATH=new_path))
>   File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 540, in check_call
>     raise CalledProcessError(retcode, cmd)
> subprocess.CalledProcessError: Command '['make', '-j1', 'kicad']' returned non-zero exit status 2

adamwolf, 2018-07-30:
Do you still have the first few lines of the output, where it prints the
summary?

(The tool should reprint that on errors, lemme add that issue.)

bson, 2018-07-30:
No, unfortunately not... don't have enough terminal history for that.  I'll kick off another one.

bson, 2018-07-31:
Here's the config output.  But I can't get it to clone ngspice now.

> build.py argument summary:
> build_type: Debug
> docs_tarball_url: http://docs.kicad-pcb.org/master/kicad-doc-HEAD.tar.gz
> footprints_ref: origin/master
> jobs: 8
> kicad_ref: origin/master
> macos_min_version: 10.13
> packages3d_ref: origin/master
> release: False
> release_name: None
> retry_failed_build: True
> symbols_ref: origin/master
> target: ['kicad']
> templates_ref: origin/master
> translations_ref: origin/master
> 
> You can change these settings.  Run ./build.py --help for details.

Looks like sourceforge is broken.

> [ 31%] Performing download step (git clone) for 'ngspice'
> cd /Users/bson/src/kicad-mac-builder/build/ngspice/src && /usr/local/Cellar/cmake/3.12.0/bin/cmake -P /Users/bson/src/kicad-mac-builder/build/ngspice/tmp/ngspice-gitclone.cmake
> Cloning into 'ngspice'...
> fatal: Could not read from remote repository.
> 
> Please make sure you have the correct access rights
> and the repository exists.
> Cloning into 'ngspice'...
> fatal: Could not read from remote repository.
> 
> Please make sure you have the correct access rights
> and the repository exists.
> Cloning into 'ngspice'...
> fatal: Could not read from remote repository.
> 
> Please make sure you have the correct access rights
> and the repository exists.
> -- Had to git clone more than once:
>           3 times.
> CMake Error at /Users/bson/src/kicad-mac-builder/build/ngspice/tmp/ngspice-gitclone.cmake:66 (message):
>   Failed to clone repository: 'git://git.code.sf.net/p/ngspice/ngspice'
> 
> make[3]: *** [ngspice/src/ngspice-stamp/ngspice-download] Error 1
> make[2]: *** [CMakeFiles/ngspice.dir/all] Error 2
> make[1]: *** [CMakeFiles/kicad.dir/rule] Error 2
> make: *** [kicad] Error 2
> Error while running make after rebuilding with a single job. Please report this issue if you cannot fix it after reading the README.
> Traceback (most recent call last):
>   File "./build.py", line 212, in <module>
>     main()
>   File "./build.py", line 208, in main
>     build(parsed_args)
>   File "./build.py", line 184, in build
>     subprocess.check_call(make_command, env=dict(os.environ, PATH=new_path))
>   File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 540, in check_call
>     raise CalledProcessError(retcode, cmd)
> subprocess.CalledProcessError: Command '['make', '-j1', 'kicad']' returned non-zero exit status 2

adamwolf, 2018-07-31:
That does sound like sourceforge is broken.  I haven't seen that here--last night's build went through just fine, here. https://builder.wayneandlayne.com/job/kicad-mac-builder-10.11-upload/49/

I ran into enough issues with sourceforge on another package that I mirrored it locally.  It isn't as straightforward to do that for git repos in CMake, unfortunately.

Can you confirm it fails for you at the command line?  If so, should we go ask the ngspice folks what to do and if they've seen this?

bson, 2018-07-31:
SF seems okay now.  But same problem, fails the same as before when trying to create the app bundle.

adamwolf, 2018-07-31:
Can you delete build/kicad-dest and build/docs?  (You shouldn't have to do
any of this, but let's see if we can get this fixed up for you.)

bson, 2018-07-31:
If I give it an empty docs dir to copy, then the build finishes successfully:

> mkdir -p ~/src/kicad-mac-builder/build/docs/share/doc/kicad/help/en

bson, 2018-07-31:
Unfortunately, it crashes with a SIGSEGV in wxAppConsoleBase::GetAppName()
Oh well, not going to mess with this anymore.

closed 2018-07-31 by bson