KiCad/kicad-mac-builder #202: Make kicad applications folder icon all fancy

adamwolf, 2018-08-02:
see https://forum.kicad.info/t/kicad-mac-folder-icon/11854

danngreen, 2018-08-03:
Hi Adam, I managed to do this with python. I have it on my repo here:
https://github.com/danngreen/kicad-mac-builder/tree/set_folder_icon

I don't know where you're at with this issue, and/or the best way to help contribute here (pull request? format-patch?), but let me know if I can do anything more.
BTW Thanks for all your work, this repo really helps osx building!

adamwolf, 2018-08-03:
A PR would be wonderful!  Thanks for the kind words.

danngreen, 2018-08-03:
Ok, PR done!

adamwolf, 2018-08-04:
We should see if we can do this with the template DMG.  I don't want to introduce a Python dependency unless I have to.  We may have to copy the contents of KiCad into it the template folder, rather than replace it.  I can take a look at this.

danngreen, 2018-08-04:
Ah, that make sense. It's not like the icon needs to change often, so doing it when preparing the template dmgs seems good.