KiCad/kicad-mac-builder #204: kicad-mac-builder-10.11-upload #53 failed

wayneandlayneci, 2018-08-04:
Build 'kicad-mac-builder-10.11-upload' is failing!

Last 50 lines of build output:

```
[...truncated 9.02 MB...]
==> default: ImportError: No module named Cocoa
==> default: Making sure any mounts are unmounted.
==> default: + cleanup
==> default: + echo 'Making sure any mounts are unmounted.'
==> default: + hdiutil detach kicad-mnt
==> default: "disk1" unmounted.
==> default: "disk1" ejected.
==> default: + hdiutil detach /Volumes/KiCad
==> default: hdiutil: detach failed - No such file or directory
==> default: + true
==> default: make[3]: *** [package-kicad-nightly/src/package-kicad-nightly-stamp/package-kicad-nightly-install] Error 1
==> default: make[2]: *** [CMakeFiles/package-kicad-nightly.dir/all] Error 2
==> default: make[1]: 
==> default: *** [CMakeFiles/package-kicad-nightly.dir/rule] Error 2
==> default: make: *** [package-kicad-nightly] Error 2
==> default: Error while running make after rebuilding with a single job. Please report this issue if you cannot fix it after reading the README.
==> default: build.py argument summary:
==> default: build_type: Debug
==> default: docs_tarball_url: http://docs.kicad-pcb.org/master/kicad-doc-HEAD.tar.gz
==> default: footprints_ref: origin/master
==> default: jobs: 1
==> default: kicad_ref: origin/master
==> default: macos_min_version: 10.11
==> default: packages3d_ref: origin/master
==> default: release: False
==> default: release_name: None
==> default: retry_failed_build: True
==> default: symbols_ref: origin/master
==> default: target: ['package-kicad-nightly', 'package-extras', 'package-kicad-unified']
==> default: templates_ref: origin/master
==> default: translations_ref: origin/master
==> default: Traceback (most recent call last):
==> default:   File "./build.py", line 218, in <module>
==> default:     main()
==> default:   File "./build.py", line 214, in main
==> default:     build(parsed_args)
==> default:   File "./build.py", line 186, in build
==> default:     subprocess.check_call(make_command, env=dict(os.environ, PATH=new_path))
==> default:   File "/usr/local/Cellar/python@2/2.7.15_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 190, in check_call
==> default:     raise CalledProcessError(retcode, cmd)
==> default: subprocess.CalledProcessError
==> default: : Command '['make', '-j1', 'package-kicad-nightly', 'package-extras', 'package-kicad-unified']' returned non-zero exit status 2
The SSH command responded with a non-zero exit status. Vagrant
assumes that this means the command failed. The output for this command
should be in the log above. Please read the output to determine what
went wrong.
Build step 'Execute shell' marked build as failure
Archiving artifacts
Recording fingerprints

```

Changes since last successful build:
- [danngreen1] aae200c2a0e0f9e113464dfe156458f25172cf1e - Folder icon made fancy in DMG (nightly and unified targets only)

[View full output](https://builder.wayneandlayne.com/job/kicad-mac-builder-10.11-upload/53/)

nickoe, 2018-08-04:
I see the same issue on my machine

adamwolf, 2018-08-04:
Reverted the fancy icon PR.  We can fix it by putting the icon in the template DMGs which gives us the same changes without adding a dependency.

wayneandlayneci, 2018-08-04:
Build was fixed!

closed 2018-08-04 by wayneandlayneci