KiCad/kicad-mac-builder #205: Add python3

nickoe, 2018-08-17:
I propose to add python3 such that we can test and verify pointhi's python3 branch on macos. I don't have macos for the time being.

nickoe, 2018-08-17:
For not it requires to disable wxpython and enabled a new build switch which is documented on his branch.

adamwolf, 2018-08-17:
I am definitely in favor of this.  I don't think I can start on this until
next Friday.

nickoe, 2018-08-23:
Although I will be away for this weekend, please keep me updated if you make any progress. :)

nickoe, 2018-08-28:
@adamwolf Did you get a chance to look into this?

adamwolf, 2018-08-28:
Yeah, I was a little side tracked with a build breakage, but it's on my
list. I want to get it started this week before any 5.0.1 stuff gets ready.