KiCad/kicad-mac-builder #217: Mojave SDK

sethhillbrand, 2018-09-28:
Hi @adamwolf 

I'm going to try to figure out what is breaking KiCad in the new Mojave this weekend unless you are already on it.

adamwolf, 2018-09-29:
What's broken in it?

adamwolf, 2018-09-29:
Sorry, lemme be a bit more verbose :)

Hi Seth!

I am on Mojave, and I can both compile KiCad using kicad-mac-builder, and I
can also run whatever version I had installed before I upgraded.

I opened KiCad.app, opened pcbnew, opened eeschema, opened some files, and
things seem fine on my end.

I haven't run the full "packaging regression test".  Before I do that, I'll
download a recent nightly and see if it works...

sethhillbrand, 2018-09-29:
Honestly, I'm not 100% clear on that yet.  We had a bug report that KiCad opened and then quickly closed when using Mojave the week before it was released.  Then there were a couple more folks on Twitter saying the same thing this week.  So this was perhaps a bad extrapolation (read: assumption!) on my part.

I just ran the upgrade overnight and am trying it out.  So far, my experience is yours.  No issues.

And don't I feel just a bit like Chicken Little. 

If nothing breaks during your regressions, I'll send an update/all clear to the list.

sethhillbrand, 2018-09-29:
Ah!  Got one.  V5.0, open eeschema, open pcbnew, switch canvas legacy->openGL->legacy a few times.  But doesn't seem to happen with nightlies.

Anything in the regression tests?

adamwolf, 2018-09-29:
Does that happen in 10.13?

I haven't been able to run the tests from the README yet, but I will be
today.

Don't worry about potentially crying wolf on this :). I much prefer crying
wolf to "let's just do it and be legends" :)

Thank you so much for helping with macOS stuff!
Adam

sethhillbrand, 2018-09-29:
I never noticed it in 10.13 but I never did this exact test.  Can't go back and check either :(

I feel like there is a "Leeroy Jenkins!" reference to be added here but perhaps that has jumped the shark (Happy Days references never go out)

sethhillbrand, 2018-09-30:
@adamwolf Spoke too soon.

Did you see https://bugs.launchpad.net/kicad/+bug/1795214 ?

I can recreate with the current 5.0 nightly from 9/28 but I can't when compiling my own under Mojave.

sethhillbrand, 2018-10-23:
Additional Mojave issue from: https://bugs.launchpad.net/kicad/+bug/1799094

sethhillbrand, 2018-10-23:
We may need to bit the bullet and release two packages for Mac.  That is what x2Go ended up doing (https://code.x2go.org/releases/binary-macosx/x2goclient/main/4.1.2.1/)

adamwolf, 2018-10-23:
I'm convinced.  I'll set up the 14.04 builder to build with 14.04 as the
minimum and upload the DMG.

adamwolf, 2018-10-29:
I have set up the 14.04 builder to build and they've been uploading for a few days.  I am going to close this, unless we see more issues.

closed 2018-10-29 by adamwolf