KiCad/kicad-mac-builder #227: Packaged Python framework is missing XML libraries

agatti, 2018-10-27:
Reproduction steps:

- Open a fully annotated schematic.
- Bring up the BOM generation dialog.
- If no plugins are present, just add one from the list of packaged plugins.
- Try to generate a report without changing the python commandline (I am using bom_html_grouped_by_value as BOM generator plugin).
- The following error occurs:

```
Run command:
python "/Applications/KiCad/kicad.app/Contents/SharedSupport/plugins/bom_html_grouped_by_value.py" "<bom_file>.xml" "<bom_path>"

Command error. Return code 1
Error messages:
Traceback (most recent call last):
  File "/Applications/KiCad/kicad.app/Contents/SharedSupport/plugins/bom_html_grouped_by_value.py", line 45, in <module>
    net = kicad_netlist_reader.netlist(sys.argv[1])
  File "/Applications/KiCad/kicad.app/Contents/SharedSupport/plugins/kicad_netlist_reader.py", line 444, in __init__
    self.load(fname)
  File "/Applications/KiCad/kicad.app/Contents/SharedSupport/plugins/kicad_netlist_reader.py", line 734, in load
    self._reader = sax.make_parser()
  File "/Applications/KiCad/kicad.app/Contents/Frameworks/Python.framework/Versions/2.7/lib/python2.7/xml/sax/__init__.py", line 93, in make_parser
    raise SAXReaderNotAvailable("No parsers found", None)
xml.sax._exceptions.SAXReaderNotAvailable: No parsers found
```

-- OS Version

ProductName:	Mac OS X
ProductVersion:	10.14
BuildVersion:	18A391

-- KiCad version

KiCad nightly 20181026-050928-ed5fa2ca2-10_14

Thank you for your time,
Alessandro

adamwolf, 2018-10-30:
This is somewhat confusing.  It seems like the bundled python may include an xml backend, but maybe  this is calling the system python, not the bundled python?

build/python-dest/Library/Frameworks/Python.framework/Versions/2.7/bin/python
Python 2.7.13 (default, Oct 30 2018, 11:23:25)
[GCC 4.2.1 Compatible Apple LLVM 10.0.0 (clang-1000.10.44.2)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import xml.sax as sax
>>> sax.make_parser()
<xml.sax.expatreader.ExpatParser instance at 0x100a79248>

Hmm.

xzcvczx, 2018-10-31:
@agatti is that when you use a download from downloads.kicad-pcb.org or when you build it yourself?

adamwolf, 2018-10-31:
Could someone who doesn't compile KiCad on their Mac test the following for
me?  Could you go into a terminal and type the following:

DYLD_PRINT_LIBRARIES=YES DYLD_PRINT_LIBRARIES_POST_LAUNCH=YES
/Applications/KiCad/kicad.app/Contents/Frameworks/Python.framework/Versions/2.7/bin/python
-c 'import xml.sax as sax; sax.make_parser()'; echo $?

and paste me the complete output?  Also, if possible and you haven't
already, could you tell me if the above steps when doing the BOM thing also
breaks for you?

Thanks a bunch, this will really help me get this solved.

xzcvczx, 2018-10-31:
```
air:qgis-relocatable simon$ DYLD_PRINT_LIBRARIES=YES DYLD_PRINT_LIBRARIES_POST_LAUNCH=YES /Applications/kicad.app/Contents/Frameworks/Python.framework/Versions/2.7/bin/python -c 'import xml.sax as sax; sax.make_parser()'; echo $?
dyld: loaded: /Applications/kicad.app/Contents/Frameworks/Python.framework/Versions/2.7/bin/python
dyld: loaded: /Applications/kicad.app/Contents/Frameworks/Python.framework/Versions/2.7/bin/../../../../Python.framework/Python
dyld: loaded: /usr/lib/libSystem.B.dylib
dyld: loaded: /usr/lib/system/libcache.dylib
dyld: loaded: /usr/lib/system/libcommonCrypto.dylib
dyld: loaded: /usr/lib/system/libcompiler_rt.dylib
dyld: loaded: /usr/lib/system/libcopyfile.dylib
dyld: loaded: /usr/lib/system/libcorecrypto.dylib
dyld: loaded: /usr/lib/system/libdispatch.dylib
dyld: loaded: /usr/lib/system/libdyld.dylib
dyld: loaded: /usr/lib/system/libkeymgr.dylib
dyld: loaded: /usr/lib/system/liblaunch.dylib
dyld: loaded: /usr/lib/system/libmacho.dylib
dyld: loaded: /usr/lib/system/libquarantine.dylib
dyld: loaded: /usr/lib/system/libremovefile.dylib
dyld: loaded: /usr/lib/system/libsystem_asl.dylib
dyld: loaded: /usr/lib/system/libsystem_blocks.dylib
dyld: loaded: /usr/lib/system/libsystem_c.dylib
dyld: loaded: /usr/lib/system/libsystem_configuration.dylib
dyld: loaded: /usr/lib/system/libsystem_coreservices.dylib
dyld: loaded: /usr/lib/system/libsystem_darwin.dylib
dyld: loaded: /usr/lib/system/libsystem_dnssd.dylib
dyld: loaded: /usr/lib/system/libsystem_info.dylib
dyld: loaded: /usr/lib/system/libsystem_m.dylib
dyld: loaded: /usr/lib/system/libsystem_malloc.dylib
dyld: loaded: /usr/lib/system/libsystem_networkextension.dylib
dyld: loaded: /usr/lib/system/libsystem_notify.dylib
dyld: loaded: /usr/lib/system/libsystem_sandbox.dylib
dyld: loaded: /usr/lib/system/libsystem_secinit.dylib
dyld: loaded: /usr/lib/system/libsystem_kernel.dylib
dyld: loaded: /usr/lib/system/libsystem_platform.dylib
dyld: loaded: /usr/lib/system/libsystem_pthread.dylib
dyld: loaded: /usr/lib/system/libsystem_symptoms.dylib
dyld: loaded: /usr/lib/system/libsystem_trace.dylib
dyld: loaded: /usr/lib/system/libunwind.dylib
dyld: loaded: /usr/lib/system/libxpc.dylib
dyld: loaded: /usr/lib/libobjc.A.dylib
dyld: loaded: /usr/lib/libc++abi.dylib
dyld: loaded: /usr/lib/libc++.1.dylib
dyld: loaded: /Applications/kicad.app/Contents/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
dyld: loaded: /Applications/kicad.app/Contents/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/../../../../../../../Python.framework/Python
dyld: loaded: /usr/lib/libSystem.B.dylib
dyld: loaded: /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
dyld: loaded: /usr/lib/system/libcache.dylib
dyld: loaded: /usr/lib/system/libcommonCrypto.dylib
dyld: loaded: /usr/lib/system/libcompiler_rt.dylib
dyld: loaded: /usr/lib/system/libcopyfile.dylib
dyld: loaded: /usr/lib/system/libcorecrypto.dylib
dyld: loaded: /usr/lib/system/libdispatch.dylib
dyld: loaded: /usr/lib/system/libdyld.dylib
dyld: loaded: /usr/lib/system/libkeymgr.dylib
dyld: loaded: /usr/lib/system/liblaunch.dylib
dyld: loaded: /usr/lib/system/libmacho.dylib
dyld: loaded: /usr/lib/system/libquarantine.dylib
dyld: loaded: /usr/lib/system/libremovefile.dylib
dyld: loaded: /usr/lib/system/libsystem_asl.dylib
dyld: loaded: /usr/lib/system/libsystem_blocks.dylib
dyld: loaded: /usr/lib/system/libsystem_c.dylib
dyld: loaded: /usr/lib/system/libsystem_configuration.dylib
dyld: loaded: /usr/lib/system/libsystem_coreservices.dylib
dyld: loaded: /usr/lib/system/libsystem_darwin.dylib
dyld: loaded: /usr/lib/system/libsystem_dnssd.dylib
dyld: loaded: /usr/lib/system/libsystem_info.dylib
dyld: loaded: /usr/lib/system/libsystem_m.dylib
dyld: loaded: /usr/lib/system/libsystem_malloc.dylib
dyld: loaded: /usr/lib/system/libsystem_networkextension.dylib
dyld: loaded: /usr/lib/system/libsystem_notify.dylib
dyld: loaded: /usr/lib/system/libsystem_sandbox.dylib
[... 149 lines cut ...]
dyld: loaded: /System/Library/Frameworks/SecurityFoundation.framework/Versions/A/SecurityFoundation
dyld: loaded: /System/Library/PrivateFrameworks/APFS.framework/Versions/A/APFS
dyld: loaded: /usr/lib/libutil.dylib
dyld: loaded: /usr/lib/libcharset.1.dylib
dyld: loaded: /System/Library/Frameworks/CoreAudio.framework/Versions/A/CoreAudio
dyld: loaded: /System/Library/Frameworks/AudioToolbox.framework/Versions/A/AudioToolbox
dyld: loaded: /System/Library/PrivateFrameworks/AppleSauce.framework/Versions/A/AppleSauce
dyld: loaded: /System/Library/PrivateFrameworks/AssertionServices.framework/Versions/A/AssertionServices
dyld: loaded: /System/Library/PrivateFrameworks/BaseBoard.framework/Versions/A/BaseBoard
dyld: loaded: /usr/lib/libmecab.1.0.0.dylib
dyld: loaded: /usr/lib/libgermantok.dylib
dyld: loaded: /usr/lib/libThaiTokenizer.dylib
dyld: loaded: /usr/lib/libChineseTokenizer.dylib
dyld: loaded: /System/Library/PrivateFrameworks/LanguageModeling.framework/Versions/A/LanguageModeling
dyld: loaded: /System/Library/PrivateFrameworks/CoreEmoji.framework/Versions/A/CoreEmoji
dyld: loaded: /System/Library/PrivateFrameworks/Lexicon.framework/Versions/A/Lexicon
dyld: loaded: /System/Library/PrivateFrameworks/LinguisticData.framework/Versions/A/LinguisticData
dyld: loaded: /usr/lib/libcmph.dylib
dyld: loaded: /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
dyld: loaded: /System/Library/Frameworks/ServiceManagement.framework/Versions/A/ServiceManagement
dyld: loaded: /System/Library/PrivateFrameworks/BackgroundTaskManagement.framework/Versions/A/BackgroundTaskManagement
dyld: loaded: /usr/lib/libxslt.1.dylib
dyld: loaded: /Applications/kicad.app/Contents/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-dynload/_io.so
0
```

xzcvczx, 2018-11-01:
Can you check your sys.paths

xzcvczx, 2018-11-01:
ah, i think i might know what the issue is.....

```
python "/Applications/kicad.app/Contents/SharedSupport/plugins/bom_html_grouped_by_value.py" "%I" "%O"
```

is that using $PATH to find/run python as such is not even using bundled python

adamwolf, 2018-11-01:
That's what I thought too! but look at the stack trace... hmm.

adamwolf, 2018-11-01:
Maybe it's using the system python, with the PYTHONPATH set from the
wrapper?  that would do it!

xzcvczx, 2018-11-01:
oh weird i wonder how thats happened, i ran those commands inside the bom gen window and i don't get the bundle paths

TheTrueTom, 2018-12-18:
Is there a workaround for this ?

adamwolf, 2018-12-18:
Maybe!  Based on the thread above, you may be able to run the BOM generator with a modified beginning.  Can you try changing the "python" part at the beginning to be:

/Applications/kicad.app/Contents/Frameworks/Python.framework/Versions/Current/bin/python

and let me know how it goes?

TheTrueTom, 2018-12-18:
> Maybe! Based on the thread above, you may be able to run the BOM generator with a modified beginning. Can you try changing the "python" part at the beginning to be:
> 
> /Applications/kicad.app/Contents/Frameworks/Python.framework/Versions/Current/bin/python
> 
> and let me know how it goes?

This is working like a charm! Thanks a lot!

adamwolf, 2018-12-18:
I dug into fixing this in KiCad, and have poked the dev mailing list when I
got stuck.  I'll run this down and get it fixed upstream.

klkl140, 2019-01-31:
the fix worked for me too. thanks

timatthehub, 2019-02-16:
I have the same problem with OS X and the new line gave me this error:

Command error. Return code -1
Error messages:
execvp(/Applications/kicad.app/Contents/Frameworks/Python.framework/Versions/Current/bin/python, /Applications/KiCad/kicad.app/Contents/SharedSupport/plugins/bom_csv_grouped_by_value.py, /Users/timblack/Documents/PAI-docs/board files/PAI-BncInputModule01/PAI-inputBoard01.xml, /Users/timblack/Documents/PAI-docs/board files/PAI-BncInputModule01/PAI-inputBoard01) failed with error 2!

adamwolf, 2019-02-20:
@timatthehub I think that's a different issue, at least by the looks of it.  That looks like there's actually an error when the script runs, which is different than this issue, which is that it isn't finding the right version.

Tyden, 2019-02-27:
@timatthehub I had the same issue. I changed the beginning to the following: 
/Applications/KiCad/kicad.app/Contents/Frameworks/Python.framework/Versions/Current/bin/python

instead of: 
/Applications/kicad.app/Contents/Frameworks/Python.framework/Versions/Current/bin/python

Mike4U, 2019-03-19:
This problem still exists on:
Application: kicad
Version: (5.1.0-0), release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.54.0 LibreSSL/2.6.4 zlib/1.2.11 nghttp2/1.24.1
Platform: Mac OS X (Darwin 18.2.0 x86_64), 64 bit, Little endian, wxMac
Build Info:
    wxWidgets: 3.0.4 (UTF-8,STL containers,compatible with 2.8)
    Boost: 1.68.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.54.0
    Compiler: Clang 9.0.0 with C++ ABI 1002

Build settings:
    USE_WX_GRAPHICS_CONTEXT=ON
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

The changing python to /Applications/KiCad/kicad.app/Contents/Frameworks/Python.framework/Versions/Current/bin/python
works on some of the scripts.

bom_csv_sorted_by_ref.py produces:
Run command:
/Applications/KiCad/kicad.app/Contents/Frameworks/Python.framework/Versions/Current/bin/python “/private/var/folders/b9/m7vp5hr925n_3r73_y07979w0000gn/T/AppTranslocation/80AE3D55-53EF-4AA1-9D50-7CAEA29C05C6/d/kicad.app/Contents/SharedSupport/plugins/bom_csv_sorted_by_ref.py” “/Users/michaelrunyan/Documents/KiCadProjects/ShiftTeensy/ShiftTeensy.xml” “/Users/michaelrunyan/Documents/KiCadProjects/ShiftTeensy/ShiftTeensy”

Success

Except the output is supposed to be ShiftTeensy.csv not ShiftTeensy

Mike4U, 2019-03-19:
Continued:
Couldn't get bom_csv_grouped_by_value.py to run:
Run command:
/Applications/KiCad/kicad.app/Contents/Frameworks/Python.framework/Versions/Current/bin/python “/private/var/folders/b9/m7vp5hr925n_3r73_y07979w0000gn/T/AppTranslocation/80AE3D55-53EF-4AA1-9D50-7CAEA29C05C6/d/kicad.app/Contents/SharedSupport/plugins/bom_csv_grouped_by_value.py” “/Users/michaelrunyan/Documents/KiCadProjects/ShiftTeensy/ShiftTeensy.xml” “/Users/michaelrunyan/Documents/KiCadProjects/ShiftTeensy/ShiftTeensy.csv”

Command error. Return code 2
Error messages:
/Applications/KiCad/kicad.app/Contents/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python: can’t open file ‘/private/var/folders/b9/m7vp5hr925n_3r73_y07979w0000gn/T/AppTranslocation/80AE3D55-53EF-4AA1-9D50-7CAEA29C05C6/d/kicad.app/Contents/SharedSupport/plugins/bom_csv_grouped_by_value.py” “/Users/michaelrunyan/Documents/KiCadProjects/ShiftTeensy/ShiftTeensy.xml” “/Users/michaelrunyan/Documents/KiCadProjects/ShiftTeensy/ShiftTeensy.csv”
‘: [Errno 2] No such file or directory

Whats with the python 2.7version, thought was supposed to be 3.x

Mike4U, 2019-03-19:
After looking around a little more something tells me I'm posting in the wrong place.  Where should I be posting this?

thanks

adamwolf, 2019-03-19:
The problem is that KiCad is just calling "python" and assuming that the
path is set correctly.

KiCad on macos still uses Python 2.7, and will until the 6 series.  The
change from 2.7 to 3 on and how it is packaged is kinda a big one, and not
super suitable for 5.1, which is primarily being released for a technical
reason regarding Linux desktops.

I need to figure out if this is better solved in KiCad or in the packaging
for macOS.

There are two workarounds posted above, I believe.  Have you tried those?

adamwolf, 2019-03-19:
I posted this upstream: https://bugs.launchpad.net/kicad/+bug/1820781

Mike4U, 2019-03-19:
Adam, thanks
Plugin   bom_csv_grouped_by_value.py    that wasn't working before now works with out the above workarounds ?
However, after quitting and restarting Kicad all the plugins became 'not available' as shown here.
[image]

The fix was to delete them and then re-add.  They then show correctly and seem to work except that the .csv extension still doesn't get added to the output file on any of the .csv plugins.  But it kinda works now...  ya

jreid27044, 2019-07-10:
The python path Tyden posted above worked for me.  Repeating here for completeness for others.

/Applications/KiCad/kicad.app/Contents/Frameworks/Python.framework/Versions/Current/bin/python