KiCad/kicad-mac-builder #230: kicad-mac-builder-10.11-5.0-upload #50 failed

wayneandlayneci, 2018-10-31:
Build 'kicad-mac-builder-10.11-5.0-upload' is failing!

Last 50 lines of build output:

```
[...truncated 69.45 KB...]
==> default: 
==> default: They will install into the site-package directory
==> default:   /usr/local/lib/python2.7/site-packages
==> default: 
==> default: See: https://docs.brew.sh/Homebrew-and-Python
==> default: ==> Summary
==> default: 🍺  /usr/local/Cellar/python@2/2.7.15_1: 7,509 files, 122.4MB, built in 7 minutes 53 seconds
==> default: ==> Installing oce dependency: tbb
==> default: ==> Downloading https://github.com/01org/tbb/archive/2019_U1.tar.gz
==> default: ==> Downloading from https://codeload.github.com/01org/tbb/tar.gz/2019_U1
==> default: ==> make tbb_build_prefix=BUILDPREFIX compiler=clang
==> default: ==> make tbb_build_prefix=BUILDPREFIX compiler=clang extra_inc=big_iron.inc
==> default: ==> python -c import setuptools... --no-user-cfg install --prefix=/usr/local/Cellar/tbb/2019_U1 --install-scripts=/usr/local/Cellar/tbb/2019_U1/bin --single-version-externally-managed --record=installed.txt
==> default: Last 15 lines from /Users/vagrant/Library/Logs/Homebrew/tbb/03.python:
==> default: swig -python -c++ -O -threads -I/usr/local/Cellar/tbb/2019_U1/include -o tbb/api_wrap.cpp tbb/api.i
==> default: creating build
==> default: creating build/temp.macosx-10.11-x86_64-2.7
==> default: creating build/temp.macosx-10.11-x86_64-2.7/tbb
==> default: clang -fno-strict-aliasing -fno-common -dynamic -g -O2 -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -I/usr/local/Cellar/tbb/2019_U1/include -I/usr/local/include -I/usr/local/opt/openssl/include -I/usr/local/opt/sqlite/include -I/usr/local/Cellar/python@2/2.7.15_1/Frameworks/Python.framework/Versions/2.7/include/python2.7 -c tbb/api_wrap.cpp -o build/temp.macosx-10.11-x86_64-2.7/tbb/api_wrap.o -std=c++11 -Wno-unused-variable
==> default: In file included from tbb/api_wrap.cpp:3137:
==> default: In file included from /usr/local/Cellar/tbb/2019_U1/include/tbb/tbb.h:81:
==> default: /usr/local/Cellar/tbb/2019_U1/include/tbb/task_group.h:132:53: error: 'uncaught_exceptions' is unavailable: introduced in macOS 10.12
==> default:             bool stack_unwinding_in_progress = std::uncaught_exceptions() > 0;
==> default:                                                     ^
==> default: /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/exception:119:63: note: 'uncaught_exceptions' has been explicitly marked unavailable here
==> default: _LIBCPP_FUNC_VIS _LIBCPP_AVAILABILITY_UNCAUGHT_EXCEPTIONS int uncaught_exceptions() _NOEXCEPT;
==> default:                                                               ^
==> default: 1 error generated.
==> default: error: command 'clang' failed with exit status 1
==> default: 
==> default: Do not report this issue to Homebrew/brew or Homebrew/core!
==> default: Error: You are using macOS 10.11.
==> default: We (and Apple) do not provide support for this old version.
==> default: You will encounter build failures and other breakages.
==> default: Please create pull-requests instead of asking for help on Homebrew's
==> default: GitHub, Discourse, Twitter or IRC. As you are running this old version,
==> default: you are responsible for resolving any issues you experience.
==> default: Error: You are using macOS 10.11.
==> default: We (and Apple) do not provide support for this old version.
==> default: You will encounter build failures and other breakages.
==> default: Please create pull-requests instead of asking for help on Homebrew's
==> default: GitHub, Discourse, Twitter or IRC. As you are running this old version,
==> default: you are responsible for resolving any issues you experience.
The SSH command responded with a non-zero exit status. Vagrant
assumes that this means the command failed. The output for this command
should be in the log above. Please read the output to determine what
went wrong.
Build step 'Execute shell' marked build as failure
Archiving artifacts

```

Changes since last successful build:
No changes

[View full output](https://builder.wayneandlayne.com/job/kicad-mac-builder-10.11-5.0-upload/50/)

nickoe, 2018-11-09:
@adamwolf Is this caused by upgrading macos?

adamwolf, 2018-11-09:
No.  I added this more than two weeks ago and this appears to be the first
failure of this.  It will automatically rebuild, and we'll see if it's a
source/dependencies thing or if it's a fluke.

adamwolf, 2018-11-09:
This is caused by the latest version of our dependencies no longer supporting macos 10.11.  I have discussed it on the mailing list and I'm going to be upgrading these to 10.12 and seeing what the backlash is from people on 10.11.

wayneandlayneci, 2018-11-13:
Build was fixed!

closed 2018-11-13 by wayneandlayneci