KiCad/kicad-mac-builder #242: libngspice bundling issue

adamwolf, 2018-11-27:
Some macOS versions are having issues with bundling libngspice.

Check to make sure they're the same versions, and compare 10.14, where it is working now, to 10.11 and 10.12 where it used to work but is now not working.

adamwolf, 2018-12-04:
This may be some sort of race condition.  It appears to randomly fail, without obvious errors, and other builds are fine.  This didn't happen *at all* for a while, but now it happens maybe 50% of the time.

I will be continuing to monitor this.

I am turning some builds to j=1, not because I think it will fix it, but because it will help me see the differences between builds that pass and builds that fail.

I am *certainly* glad that I have the checker that checks every single file in the bundle!

adamwolf, 2018-12-16:
This hasn't been seen since switching to one job.  After we get a few nights of good nightlies, I'll do more tests.

nickoe, 2019-01-03:
Any update? How can I check if there is an issue?

adamwolf, 2019-01-03:
It will error out during the build since there is a tester script.  I have
set the builds to one core and now don't see it, but I would love to
troubleshoot or have someone else troubleshoot it.  It didn't show up 100%.

adamwolf, 2019-01-10:
This is still happening intermittently during builds.  I am investigating.  We *explicitly* fixup the libngspice library that it complains about.  The next build that fails, I need to make sure that NGSPICE_HELPER is set in the CMakeLists.txt for kicad.

adamwolf, 2019-01-10:
I am continuing to dig into this today.

adamwolf, 2019-07-01:
This no longer happens.

closed 2019-07-01 by adamwolf