KiCad/kicad-mac-builder #254: kicad-mac-builder-10.14-upload #46 failed

wayneandlayneci, 2018-12-08:
Build 'kicad-mac-builder-10.14-upload' is failing!

Last 50 lines of build output:

```
[...truncated 1.64 KB...]
+ vagrant halt
+ PATH=/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/bin
+ vagrant destroy -f
==> default: Destroying VM and associated drives...
+ rm -rf .vagrant
+ PATH=/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/bin
+ vagrant up
Bringing machine 'default' up with 'virtualbox' provider...
==> default: Importing base box 'macos_10.14'...
[KProgress: 10%[KProgress: 20%[KProgress: 30%[KProgress: 40%[KProgress: 90%[K==> default: Matching MAC address for NAT networking...
==> default: Setting the name of the VM: 1014_default_1544275552642_62801
==> default: Fixed port collision for 22 => 2222. Now on port 2204.
==> default: Vagrant has detected a configuration issue which exposes a
==> default: vulnerability with the installed version of VirtualBox. The
==> default: current guest is configured to use an E1000 NIC type for a
==> default: network adapter which is vulnerable in this version of VirtualBox.
==> default: Ensure the guest is trusted to use this configuration or update
==> default: the NIC type using one of the methods below:
==> default: 
==> default:   https://www.vagrantup.com/docs/virtualbox/configuration.html#default-nic-type
==> default:   https://www.vagrantup.com/docs/virtualbox/networking.html#virtualbox-nic-type
==> default: Clearing any previously set network interfaces...
==> default: Preparing network interfaces based on configuration...
    default: Adapter 1: nat
==> default: Forwarding ports...
    default: 22 (guest) => 2204 (host) (adapter 1)
==> default: Booting VM...
==> default: Waiting for machine to boot. This may take a few minutes...
    default: SSH address: 127.0.0.1:2204
    default: SSH username: vagrant
    default: SSH auth method: private key
Timed out while waiting for the machine to boot. This means that
Vagrant was unable to communicate with the guest machine within
the configured ("config.vm.boot_timeout" value) time period.

If you look above, you should be able to see the error(s) that
Vagrant had when attempting to connect to the machine. These errors
are usually good hints as to what may be wrong.

If you're using a custom box, make sure that networking is properly
working and you're able to connect to the machine. It is a common
problem that networking isn't setup properly in these boxes.
Verify that authentication configurations are also setup properly,
as well.

If the box appears to be booting properly, you may want to increase
the timeout ("config.vm.boot_timeout") value.
Build step 'Execute shell' marked build as failure
Archiving artifacts

```

Changes since last successful build:
- [adamwolf] fd9bf2f7cdf550c0d2aa9ee21599b5049bac29f9 - Temporarily moving to 1 job to debug libngspice build issue.

- [adamwolf] 83736150c4dd15e7f54b28645225e8f40ba8043b - No longer version pinning cmake.

- [adamwolf] 38d8749504c6da56fe7b4e4dcb6b08c41ac81407 - Add cmake to dependencies list after removing version pinned version.

[View full output](https://builder.wayneandlayne.com/job/kicad-mac-builder-10.14-upload/46/)

xzcvczx, 2018-12-14:
is this the reason there have been no nightlies since 29/11?

adamwolf, 2018-12-14:
The first issue is a race condition with building the package that started
in mid November.  It has to do with linking ngspice.  When I move things to
only build with one thread at a time, this goes away--which isn't fixing
the race, but is a workaround.  A few days after doing that, many of the
jobs started failing on uploading DMGs to kicad-pcb via ssh--which makes no
sense, as I can do it just fine with the same key, and the upload portion
of the job hasn't changed since 5 came out.  I suspect it has to do with
how long the uploads take on that connection, and something in the router.
Furthermore, it now appears that vagrant is timing out starting the VMs.

I'm frustrated without how needlessly fragile building things for macOS
is.  When I don't build in VMs, the box needs to become dedicated to *just*
KiCad, which is a bummer.  When I build in VMs, I spend ~4 hours a month or
more just keeping the builds up--and then I get weird failures that are
caused by... nothing?  Harrumph.

adamwolf, 2018-12-14:
yeah, this is incredibly frustrating.  There are successful builds for all the jobs within the last few days--but all of them failed uploading the DMGs.

https://builder.wayneandlayne.com/job/kicad-mac-builder-10.14-upload/43/console
https://builder.wayneandlayne.com/job/kicad-mac-builder-10.11-master-upload/170/console
https://builder.wayneandlayne.com/job/kicad-mac-builder-10.11-5.0-upload/86/console

xzcvczx, 2018-12-14:
admittedly half the reason for commenting on the issue (despite not really thinking it was the fault) was i was unsure whether you were aware that nightlies had failed

adamwolf, 2018-12-14:
Sorry if I seem frustrated at you--it's definitely not you :)

adamwolf, 2018-12-16:
Fixed the ssh issue, fixed the jobs dying "for no reason" (switching to 1 core for building means my timeout was too short...), fixed the GLM version issue, and tada!  nightlies are back!

closed 2018-12-16 by adamwolf

xzcvczx, 2018-12-16:
thanks adam, btw did you know the nightlies were broke and you were just too busy to deal with it? or were you unaware?

adamwolf, 2018-12-16:
I did know, but the builds were taking *six hours* so I could only do so
many iterations per day.  That isn't to say that when you poked me so
didn't spend another half hour on it :p