KiCad/kicad-mac-builder #257: update ngspice

adamwolf, 2019-01-05:
@nickoe made a PR #256 for updating to the latest ngspice tag.  It didn't work, and fails with:

``
03:08:47     default: [ 57%] Building CXX object eeschema/CMakeFiles/eeschema_kiface.dir/sim/ngspice.cpp.o
03:08:48     default: In file included from /vagrant/build/kicad/src/kicad/eeschema/sim/ngspice.cpp:28:
03:08:48     default: /vagrant/build/kicad/src/kicad/eeschema/sim/ngspice.h:31:10: fatal error: 'ngspice/sharedspice.h' file not found
03:08:48     default: #include <ngspice/sharedspice.h>
03:08:48     default:          ^
03:08:49     default: 1 error generated.
03:08:49     default: make[6]: *** [eeschema/CMakeFiles/eeschema_kiface.dir/sim/ngspice.cpp.o] Error 1
``

Let's fix this.

nickoe, 2019-01-05:
Where is the full log?

adamwolf, 2019-06-24:
@nickoe I tried updating ngspice again.  Here's the log.

https://builder.wayneandlayne.com/job/kicad-mac-builder-10.14-upload/249/consoleFull

nickoe, 2019-06-24:
This is the error:
```
[ 53%] Building CXX object eeschema/CMakeFiles/eeschema_kiface_objects.dir/sim/ngspice.cpp.o
In file included from /vagrant/build/kicad/src/kicad/eeschema/sim/ngspice.cpp:28:
/vagrant/build/kicad/src/kicad/eeschema/sim/ngspice.h:31:10: fatal error: 'ngspice/sharedspice.h' file not found
#include <ngspice/sharedspice.h>
         ^~~~~~~~~~~~~~~~~~~~~~~
1 error generated.

```

adamwolf, 2019-06-24:
Yeah... that didn't happen with ngspice 28.  I'm going to revert it but
keep it building on a branch.  I am not able to put more effort into the
ngspice stuff at this time but maybe someone else can do PRs then.

adamwolf, 2019-06-24:
The job will be at https://builder.wayneandlayne.com/job/kicad-mac-builder-10.14-ngspice-upload/

adamwolf, 2019-06-25:
Alright, the job built, here: https://builder.wayneandlayne.com/job/kicad-mac-builder-10.14-ngspice-upload/1/consoleFull

adamwolf, 2019-06-25:
nickoe, the build is on a completely brand new VM every single time.  I'll try changing the include path.  Thanks for the sleuthing.

adamwolf, 2019-06-27:
I got it working on my dev machine.  Pushed to the branch, we 'll see what happens overnight.

nickoe, 2019-07-01:
@adamwolf Could you check your build nodes? The builds have been running for 1 day and 15 hours by now. Maybe the build script is running too many things in parralel and using all the machine memory?

adamwolf, 2019-07-01:
I'll take a look. Thanks Nick.  I was trying to get the build machines to
also have build a new template 10.15 Beta VM so it's likely I was just
optimistic...

nickoe, 2019-07-01:
@adamwolf ok, the build did became green!!!! woo

https://builder.wayneandlayne.com/job/kicad-mac-builder-10.14-ngspice-upload/10/

I guess we can merge https://github.com/KiCad/kicad-mac-builder/pull/286

adamwolf, 2019-07-01:
Thanks nick and everyone for your patience!

closed 2019-07-01 by adamwolf

nickoe, 2019-07-01:
@adamwolf Maybe you should delete the ngspice-upload build and re-enable https://builder.wayneandlayne.com/job/kicad-mac-builder-10.14-upload/  ?

adamwolf, 2019-07-02:
Done!

PR #286 (2019-07-01): Ngspice upgrade again again agian
Fixes #257