KiCad/kicad-mac-builder #274: Can't run sub-applications directly

sethhillbrand, 2019-02-13:
Using the OSX 5.0.2-5_10.14 download, you cannot directly execute the sub-applications eeschema/pcbnew/etc due to a missing library error.

[image]

adamwolf, 2019-02-13:
Please confirm you are not running out of a disk image, and please confirm
you have run the main kicad.app at least once since installing it.

theodoreren, 2019-02-13:
It was my fault. Now I am able to open pcb files and schematic files. But I have another issue. I am not sure why the schematic viewer keeps crashing. I have attached the error report here.

"
Process:               eeschema [959]
Path:                  /Volumes/VOLUME/*/kicad.app/Contents/Applications/eeschema.app/Contents/MacOS/eeschema
Identifier:            org.kicad-pcb.eeschema
Version:               ??? (???)
Code Type:             X86-64 (Native)
Parent Process:        kicad [951]
Responsible:           eeschema [959]
User ID:               501

Date/Time:             2019-02-12 23:14:29.196 -0500
OS Version:            Mac OS X 10.14 (18A391)
Report Version:        12
Anonymous UUID:        187AA4E9-6884-839B-E4FC-6AFDEA5AE1BB

Time Awake Since Boot: 630 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x000007f83c2e2ef0
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [959]

VM Regions Near 0x7f83c2e2ef0:
    CoreAnimation          0000000139779000-000000013991f000 [ 1688K] rw-/rwx SM=PRV  
--> 
    STACK GUARD            000070000939d000-000070000939e000 [    4K] ---/rwx SM=NUL  stack guard for thread 1

Application Specific Information:
objc_msgSend() selector name: _subclassOverridesDrawBackgroundInRect

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libobjc.A.dylib               	0x00007fff5c4d3a1d objc_msgSend + 29
1   com.apple.AppKit              	0x00007fff2de1e339 -[NSTableBackgroundView drawRect:] + 274
2   com.apple.AppKit              	0x00007fff2db9c205 _NSViewDrawRect + 66
3   com.apple.AppKit              	0x00007fff2db9aabd -[NSView(NSInternal) _recursive:displayRectIgnoringOpacity:inContext:shouldChangeFontReferenceColor:stopAtLayerBackedViews:] + 1545
4   com.apple.AppKit              	0x00007fff2db9a4a2 __46-[NSView(NSLayerKitGlue) drawLayer:inContext:]_block_invoke + 192
5   com.apple.AppKit              	0x00007fff2db9a201 -[NSView(NSLayerKitGlue) _drawViewBackingLayer:inContext:drawingHandler:] + 1769
6   com.apple.AppKit              	0x00007fff2dc701a8 -[_NSBackingLayerContents drawLayer:inContext:] + 165
7   com.apple.QuartzCore          	0x00007fff3b4c1d56 -[CALayer drawInContext:] + 281
8   com.apple.AppKit              	0x00007fff2dc6fe02 -[_NSTiledLayer drawTile:inContext:] + 605
9   com.apple.AppKit              	0x00007fff2dc6fb4b -[_NSTiledLayerContents drawLayer:inContext:] + 181
10  com.apple.QuartzCore          	0x00007fff3b4c1d56 -[CALayer drawInContext:] + 281
11  com.apple.AppKit              	0x00007fff2dc6fa8b -[NSTileLayer drawInContext:] + 148
12  com.apple.QuartzCore          	0x00007fff3b4aeaaf CABackingStoreUpdate_ + 577
13  com.apple.QuartzCore          	0x00007fff3b510325 invocation function for block in CA::Layer::display_() + 53
14  com.apple.QuartzCore          	0x00007fff3b4adc90 -[CALayer _display] + 1839
15  com.apple.AppKit              	0x00007fff2dc6f9d3 -[NSTileLayer display] + 114
16  com.apple.AppKit              	0x00007fff2dc6de92 -[_NSTiledLayerContents update:shouldCallPrepareContent:] + 4780
17  com.apple.AppKit              	0x00007fff2dc6c952 -[_NSTiledLayer display] + 411
18  com.apple.QuartzCore          	0x00007fff3b4ad1bc CA::Layer::display_if_needed(CA::Transaction*) + 634
19  com.apple.QuartzCore          	0x00007fff3b49b447 CA::Context::commit_transaction(CA::Transaction*) + 319
20  com.apple.QuartzCore          	0x00007fff3b49ad20 CA::Transaction::commit() + 576
21  com.apple.AppKit              	0x00007fff2db748f5 __65+[CATransaction(NSCATransaction) NS_setFlushesWithDisplayRefresh]_block_invoke + 274
22  com.apple.CoreFoundation      	0x00007fff3057895d __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
23  com.apple.CoreFoundation      	0x00007fff30578892 __CFRunLoopDoObservers + 452
24  com.apple.CoreFoundation      	0x00007fff3051a3c5 __CFRunLoopRun + 1166
25  com.apple.CoreFoundation      	0x00007fff30519ce4 CFRunLoopRunSpecific + 463
26  com.apple.HIToolbox           	0x00007fff2f7b3895 RunCurrentEventLoopInMode + 293
27  com.apple.HIToolbox           	0x00007fff2f7b34d4 ReceiveNextEventCommon + 371
[... 63 lines cut ...]
1   com.apple.CoreFoundation      	0x00007fff30548c83 __CFSocketManager + 639
2   libsystem_pthread.dylib       	0x00007fff5d7a233d _pthread_body + 126
3   libsystem_pthread.dylib       	0x00007fff5d7a52a7 _pthread_start + 70
4   libsystem_pthread.dylib       	0x00007fff5d7a1425 thread_start + 13

Thread 7:
0   libsystem_pthread.dylib       	0x00007fff5d7a1408 start_wqthread + 0
1   ???                           	0x000070000972fc10 0 + 123145460841488

Thread 8:
0   libsystem_pthread.dylib       	0x00007fff5d7a1408 start_wqthread + 0
1   ???                           	0x0000000000001303 0 + 4867

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x00000000000000a8  rbx: 0x00000000000000b0  rcx: 0x0000000000000380  rdx: 0x00007fff8f2c1c40
  rdi: 0x00007f83c2e2e3b0  rsi: 0x00007fff2e6ef31f  rbp: 0x00007ffee7bc63b0  rsp: 0x00007ffee7bc6338
   r8: 0x0000000000000000   r9: 0x0000000000000005  r10: 0x000007f83c2e2ed8  r11: 0x00007fff2e6ef31f
  r12: 0x00007f83c22d1770  r13: 0x0000000000000001  r14: 0x00007fff2e65da59  r15: 0x00007ffee7bc63c0
  rip: 0x00007fff5c4d3a1d  rfl: 0x0000000000010206  cr2: 0x000007f83c2e2ef0
  
Logical CPU:     0
Error Code:      0x00000004
Trap Number:     14

"

sethhillbrand, 2019-02-13:
Thank you for confirming.  This issue will be closed.  We will need to deal with the crash in a separate report.

closed 2019-02-13 by sethhillbrand