KiCad/kicad-mac-builder #282: Build fails on 10.14

hedefalk, 2019-05-17:
Sorry to post as a bug if I'm missing some step. Just reading the readme and installed the brew deps and then

    ./build.py --target package-kicad-nightly

I get:

> wxPython/Phoenix does not appear to be installed on the system

Tail of build log:

```
-- Python module install path: /Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad/src/kicad-build/kicad/kicad.app/Contents/Frameworks/python/site-packages
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/wxwidgets-dest/lib/python2.7/site-packages/wx-3.0-osx_cocoa/wx/__init__.py", line 45, in <module>
    from wx._core import *
  File "/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/wxwidgets-dest/lib/python2.7/site-packages/wx-3.0-osx_cocoa/wx/_core.py", line 4, in <module>
    import _core_
ImportError: dlopen(/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/wxwidgets-dest/lib/python2.7/site-packages/wx-3.0-osx_cocoa/wx/_core_.so, 2): Symbol not found: __ZN22wxStringOperationsUtf816ms_utf8IterTableE
  Referenced from: /Users/viktor/dev/projects/kicad/kicad-mac-builder/build/wxwidgets-dest/lib/python2.7/site-packages/wx-3.0-osx_cocoa/wx/_core_.so
  Expected in: /Users/viktor/dev/projects/kicad/kicad-mac-builder/build/wxwidgets-dest/lib/libwx_osx_cocoau-3.0.0.dylib
 in /Users/viktor/dev/projects/kicad/kicad-mac-builder/build/wxwidgets-dest/lib/python2.7/site-packages/wx-3.0-osx_cocoa/wx/_core_.so
CMake Error at CMakeModules/FindwxPython.cmake:52 (message):
  wxPython/Phoenix does not appear to be installed on the system
Call Stack (most recent call first):
  CMakeLists.txt:710 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad/src/kicad-build/CMakeFiles/CMakeOutput.log".
See also "/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad/src/kicad-build/CMakeFiles/CMakeError.log".
make[3]: *** [kicad/src/kicad-stamp/kicad-configure] Error 1
make[2]: *** [CMakeFiles/kicad.dir/all] Error 2
make[1]: *** [CMakeFiles/package-kicad-nightly.dir/rule] Error 2
make: *** [package-kicad-nightly] Error 2
Error while running make after rebuilding with a single job. Please report this issue if you cannot fix it after reading the README.
build.py argument summary:
build_type: Debug
docs_tarball_url: http://docs.kicad-pcb.org/kicad-doc-HEAD.tar.gz
extra_version: None
footprints_ref: origin/master
jobs: 1
kicad_ref: origin/master
macos_min_version: 10.14
packages3d_ref: origin/master
release: False
release_name: None
retry_failed_build: True
symbols_ref: origin/master
target: ['package-kicad-nightly']
templates_ref: origin/master
translations_ref: origin/master
Traceback (most recent call last):
  File "./build.py", line 223, in <module>
    main()
  File "./build.py", line 219, in main
    build(parsed_args)
  File "./build.py", line 191, in build
    subprocess.check_call(make_command, env=dict(os.environ, PATH=new_path))
  File "/usr/local/Cellar/python@2/2.7.15_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '-j1', 'package-kicad-nightly']' returned non-zero exit status 2
```

hedefalk, 2019-05-17:
Is this possibly the error: 

>ImportError: dlopen(/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/wxwidgets-dest/lib/python2.7/site-packages/wx-3.0-osx_cocoa/wx/_core_.so, 2): Symbol not found: __ZN22wxStringOperationsUtf816ms_utf8IterTableE

adamwolf, 2019-05-17:
That's a new one!  Our build machines build on 10.11 and 10.14 every night, but I know it is not on 10.14.4.  They install from scratch every night, so they download the latest versions of all the dependencies, just like if a new user was starting.

That is an error I have never seen.  I just did a rebuild on my dev laptop, which is running 10.14.4, and I did not have this issue.

I can investigate, but it may be a few days.  Have you tried deleting your build directory and trying again?  I know that sounds silly but it is worth a try.

hedefalk, 2019-05-17:
@adamwolf Yey, rimrafed build and that was definitely an improvement! I think I tried this build about 6 month ago or something so that might have been the issue…

Build took about an hour before failing now with something I think might be a dupe of another issue here:

```
[ 90%] Building CXX object plugins/3d/oce/CMakeFiles/s3d_plugin_oce.dir/loadmodel.cpp.o
make[5]: *** No rule to make target `/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/System/Library/Frameworks/OpenGL.framework', needed by `kicad/kicad.app/Contents/PlugIns/3d/libs3d_plugin_oce.so'.  Stop.
make[4]: *** [plugins/3d/oce/CMakeFiles/s3d_plugin_oce.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [kicad/src/kicad-stamp/kicad-build] Error 2
make[1]: *** [CMakeFiles/kicad.dir/all] Error 2
make: *** [all] Error 2
Error while running make after rebuilding with a single job. Please report this issue if you cannot fix it after reading the README.
build.py argument summary:
build_type: Debug
docs_tarball_url: http://docs.kicad-pcb.org/kicad-doc-HEAD.tar.gz
extra_version: None
footprints_ref: origin/master
jobs: 1
kicad_ref: origin/master
macos_min_version: 10.14
packages3d_ref: origin/master
release: False
release_name: None
retry_failed_build: True
symbols_ref: origin/master
target: []
templates_ref: origin/master
translations_ref: origin/master
Traceback (most recent call last):
  File "./build.py", line 223, in <module>
    main()
  File "./build.py", line 219, in main
    build(parsed_args)
  File "./build.py", line 191, in build
    subprocess.check_call(make_command, env=dict(os.environ, PATH=new_path))
  File "/usr/local/Cellar/python@2/2.7.15_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '-j1']' returned non-zero exit status 2
```

adamwolf, 2019-05-17:
This is one I can solve :)

The Homebrew OCE bottle is broken. It refers to the SDK for 10.12. You
have two options, both of which are commands you run in the terminal.

You can either build it yourself with: brew install
--build-from-source brewsci/science/oce or use the bottle I made and
include in this repository. brew install -f
external/oce-0.18.2.el_capitan.bottle.1.tar.gz

Let me know how it goes!  Feel free to close this issue if these
instructions work and you get a build!

hedefalk, 2019-05-17:
@adamwolf Thanks!

Sorry for reporting back so slowly, but it's actually the build that takes quite some time and I don't want to flood the issue with "gonna try that" messages :)

So I came a lot further after building OCE from source. But still failing with _something_:

```
[ 57%] Checking that all loader dependencies are system-provided or relative
/Users/viktor/dev/projects/kicad/kicad-mac-builder/kicad-mac-builder/bin/verify-app.sh /Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app
Checking /Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app for non-relative and non-system references.
	/opt/local/lib/db48/libdb-4.8.dylib (compatibility version 0.0.0, current version 0.0.0)
Issues found.
References with issues are indented.
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/MacOS/dxf2idf:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/MacOS/idf2vrml:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/MacOS/kicad:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/MacOS/kicad2step:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/MacOS/idfrect:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/MacOS/idfcyl:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/MacOS/kicad-ogltest:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/PlugIns/3d/libs3d_plugin_vrml.so:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/PlugIns/3d/libs3d_plugin_idf.so:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/PlugIns/3d/libs3d_plugin_oce.so:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/PlugIns/_eeschema.kiface:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/PlugIns/_cvpcb.kiface:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/PlugIns/_gerbview.kiface:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/PlugIns/_pcb_calculator.kiface:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/PlugIns/_pcbnew.kiface:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/PlugIns/_pl_editor.kiface:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/PlugIns/sim/libngspice.0.dylib:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/PlugIns/sim/ngspice/spice2poly.cm:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/PlugIns/sim/ngspice/digital.cm:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/PlugIns/sim/ngspice/xtraevt.cm:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/PlugIns/sim/ngspice/xtradev.cm:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/PlugIns/sim/ngspice/analog.cm:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/Applications/eeschema.app/Contents/MacOS/eeschema:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/Applications/gerbview.app/Contents/MacOS/gerbview:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/Applications/pcbnew.app/Contents/MacOS/pcbnew:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/Applications/bitmap2component.app/Contents/MacOS/bitmap2component:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/Applications/pcb_calculator.app/Contents/MacOS/pcb_calculator:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/Applications/pl_editor.app/Contents/MacOS/pl_editor:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/Frameworks/libTKXCAF.11.0.0.dylib:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/Frameworks/libTKernel.11.0.0.dylib:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/Frameworks/libTKV3d.11.0.0.dylib:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/Frameworks/libTKHLR.11.0.0.dylib:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/Frameworks/libwx_osx_cocoau_gl-3.0.0.4.0.dylib:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/Frameworks/libcairo.2.dylib:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/Frameworks/libTKPCAF.11.0.0.dylib:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/Frameworks/libTKPShape.11.0.0.dylib:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/Frameworks/libTKBRep.11.0.0.dylib:
/Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app/Contents/Frameworks/python/site-packages/wx-3.0-osx_cocoa/wx/_calendar.so:
[... cut ...]
  File "./build.py", line 223, in <module>
    main()
  File "./build.py", line 219, in main
    build(parsed_args)
  File "./build.py", line 191, in build
    subprocess.check_call(make_command, env=dict(os.environ, PATH=new_path))
  File "/usr/local/Cellar/python@2/2.7.15_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '-j1']' returned non-zero exit status 2
```

hedefalk, 2019-05-17:
so libdb-4.8.dylib is for some reason `/opt/local/lib/db48/libdb-4.8.dylib (compatibility version 0.0.0, current version 0.0.0)` and which is not a relative path and this i bad?

adamwolf, 2019-05-17:
Yeah.  For a dmg to be useful to other people, kicad.app needs to only refer to libraries included inside of it, or ones that come with macOS.

So, there's a few solutions to this.

1) If you want to create dmgs or binaries you give to other folks, we work through this and figure out why this is happening (I don't see it on my builds, but I trust that it's happening to you) :)
2) If you want to build and its dependencies, run ./build.sh <whatever options you want> kicad instead of the package targets.

Regardless, I'll keep this as a bug and look into it, but depending on what you're doing, you may be able to work around it real quickly.

hedefalk, 2019-05-20:
@adamwolf Thanks! For now I'm fine with the binaries to run locally. Great!

I guess the external dep might be some homebrew related thing?

I'll close this ticket if fine by you! Let me know if I should open another one for libdb-4.8.dylib.

Thanks a lot!

hedefalk, 2019-05-20:
Oh, if it wasn't obvious: `./build.sh kicad ` worked :)

closed 2019-05-20 by hedefalk

adamwolf, 2019-05-20:
Thanks! If you could make a new ticket with just the libdb error (and maybe
refer to this one), that would be great. I'll check back in a few days and
do it myself if you're too busy.

Thanks for your help!