KiCad/kicad-mac-builder #283: Absolute dependency to libdb-4.8.dylib failing dmg build on 10.14.4

hedefalk, 2019-05-20:
```
[ 57%] Checking that all loader dependencies are system-provided or relative
/Users/viktor/dev/projects/kicad/kicad-mac-builder/kicad-mac-builder/bin/verify-app.sh /Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app
Checking /Users/viktor/dev/projects/kicad/kicad-mac-builder/build/kicad-dest/kicad.app for non-relative and non-system references.
	/opt/local/lib/db48/libdb-4.8.dylib (compatibility version 0.0.0, current version 0.0.0)
Issues found.

```

https://github.com/KiCad/kicad-mac-builder/issues/282

adamwolf, 2019-05-20:
What else do you have in /opt/local?  Did you install homebrew there?

hedefalk, 2019-05-21:
Hm, it's actually probably MacPorts… Think I installed that recently for some reason.

hedefalk, 2019-05-21:
```
~ ❯❯❯ ll /opt/local/
total 0
drwxr-xr-x   13 root  wheel   416B May  5 11:18 ./
drwxr-xr-x    5 root  wheel   160B May  4 20:45 ../
drwxr-xr-x    3 root  wheel    96B May  5 11:08 Library/
drwxr-xr-x  632 root  wheel    20K May  5 11:51 bin/
drwxr-xr-x   14 root  wheel   448B May  5 11:48 etc/
drwxr-xr-x  192 root  wheel   6.0K May  5 11:35 include/
drwxr-xr-x  706 root  wheel    22K May  5 11:51 lib/
drwxr-xr-x   15 root  wheel   480B May  5 11:48 libexec/
lrwxr-xr-x    1 root  wheel     9B May  5 11:05 man@ -> share/man
drwxr-xr-x    9 root  wheel   288B May  5 11:08 sbin/
drwxr-xr-x   60 root  wheel   1.9K May  5 11:43 share/
drwxr-xr-x    4 root  admin   128B May  5 11:18 src/
drwxr-xr-x    8 root  wheel   256B May  5 11:18 var/
```

hedefalk, 2019-05-21:
There's lots there:

```
~ ❯❯❯ ll /opt/local/lib
total 460848
drwxr-xr-x  706 root  wheel    22K May  5 11:51 ./
drwxr-xr-x   13 root  wheel   416B May  5 11:18 ../
-rw-r--r--    1 root  wheel     0B Oct  3  2018 .turd_MacPorts
drwxr-xr-x    4 root  admin   128B May  5 11:13 ImageMagick-6.9.9/
drwxr-xr-x    3 root  admin    96B May  5 11:51 X11/
drwxr-xr-x   61 root  admin   1.9K May  5 11:13 aspell-0.60/
drwxr-xr-x    4 root  admin   128B May  5 11:18 atkmm-1.6/
drwxr-xr-x    8 root  admin   256B May  5 11:18 cairo/
drwxr-xr-x    3 root  admin    96B May  5 11:18 cairomm-1.0/
drwxr-xr-x    7 root  admin   224B May  5 11:18 cmake/
drwxr-xr-x   12 root  admin   384B May  5 11:07 db48/
drwxr-xr-x    3 root  admin    96B May  5 11:18 dbus-1.0/
drwxr-xr-x    4 root  admin   128B May  5 11:19 enchant/
drwxr-xr-x   14 root  admin   448B May  5 11:08 engines/
drwxr-xr-x   47 root  admin   1.5K May  5 11:25 gcc8/
drwxr-xr-x    3 root  admin    96B May  5 11:18 gdk-pixbuf-2.0/
drwxr-xr-x    3 root  admin    96B May  5 11:19 gdkmm-2.4/
drwxr-xr-x    7 root  admin   224B May  5 11:07 gettext/
drwxr-xr-x    3 root  admin    96B May  5 11:18 giomm-2.4/
drwxr-xr-x   29 root  admin   928B May  5 11:20 girepository-1.0/
drwxr-xr-x    3 root  admin    96B May  5 11:18 glib-2.0/
drwxr-xr-x    4 root  admin   128B May  5 11:18 glibmm-2.4/
drwxr-xr-x    3 root  admin    96B May  5 11:18 gobject-introspection/
drwxr-xr-x    5 root  admin   160B May  5 11:19 gtk-2.0/
drwxr-xr-x    4 root  admin   128B May  5 11:19 gtkmm-2.4/
drwxr-xr-x    6 root  admin   192B May  5 11:08 icu/
drwxr-xr-x   14 root  admin   448B May  5 11:43 inkscape/
-rwxr-xr-x    1 root  admin    18K Oct 25  2018 libAppleWM.7.dylib*
-rw-r--r--    1 root  admin   8.7K Oct 25  2018 libAppleWM.a
lrwxr-xr-x    1 root  admin    18B Oct 25  2018 libAppleWM.dylib@ -> libAppleWM.7.dylib
………
```

hedefalk, 2019-05-21:
Might have been inkscape I needed it for…

om2kw, 2019-05-24:
can confirm the same problem, it occurs during packaging so
build.py --target kicad works ok

adamwolf, 2019-05-24:
Do you also have macports installed?

om2kw, 2019-05-24:
yes, and also inkscape as was mentioned above

adamwolf, 2019-05-25:
Hmm.  Interesting!  I'm headed out for a holiday weekend here in the US but
I'll bet this is something that we can easily clear up.