KiCad/kicad-mac-builder #285: VBoxManage: error: Appliance import failed

nickoe, 2019-06-21:
Hello Adam, 

It looks like your builder is failing since the 10th.

```
+ vagrant up
Bringing machine 'default' up with 'virtualbox' provider...
==> default: Importing base box 'macos_10.14'...

[KProgress: 90%There was an error while executing `VBoxManage`, a CLI used by Vagrant
for controlling VirtualBox. The command and stderr is shown below.

Command: ["import", "/Users/jenkinsbuilder/.vagrant.d/boxes/macos_10.14/0/virtualbox/box.ovf", "--vsys", "0", "--vmname", "mojave_default_1539268301096_27948_1561111224594_21598", "--vsys", "0", "--unit", "11", "--disk", "/Users/jenkinsbuilder/VirtualBox VMs/mojave_default_1539268301096_27948_1561111224594_21598/box-disk001.vmdk"]

Stderr: 0%...10%...20%...30%...40%...50%...60%...70%...80%...90%...100%
Interpreting /Users/jenkinsbuilder/.vagrant.d/boxes/macos_10.14/0/virtualbox/box.ovf...
OK.
0%...
Progress state: NS_ERROR_INVALID_ARG
VBoxManage: error: Appliance import failed
VBoxManage: error: Code NS_ERROR_INVALID_ARG (0x80070057) - Invalid argument value (extended info not available)
VBoxManage: error: Context: "RTEXITCODE handleImportAppliance(HandlerArg *)" at line 957 of file VBoxManageAppliance.cpp
Build step 'Execute shell' marked build as failure
```

adamwolf, 2019-06-24:
Yeah, I fixed it.  Now it's complaining about the ngspice 30 upgrade.

closed 2019-06-24 by adamwolf

nickoe, 2019-06-24:
Ok, cool, I will check your new logs later today.