KiCad/kicad-mac-builder #288: macOS Catalina

a7hybnj2, 2019-08-07:
I run nightlies and have been itching to upgrade to Catalina but have held off because of these issues I found (these are not my original reports).

https://bugs.launchpad.net/kicad/+bug/1834717
https://forum.kicad.info/t/program-doesnt-run-on-macos-catalina-10-15-19a487m/17806

I am hoping to just get confirmation from anyone that has tried KiCad with Catalina and what will be needed to make it work.

adamwolf, 2019-08-07:
Once we get 5.1.4 out the door I will be taking a look at this.  If you
have a Catalina machine and want to help or want to test, let me know.

a7hybnj2, 2019-08-07:
@adamwolf I will be happy to be a tester/bug reporter. I have been running nightlies and reporting issues for some time now.

Let me know when a build is stable enough to open on Catalina and I will upgrade to it.

adamwolf, 2019-08-29:
I now have a Catalina vagrant box, so I can start working on this.

a7hybnj2, 2019-08-29:
Great news! Let me know when it is stable enough and I will go Catalina(beta).

adamwolf, 2019-09-26:
Heads up, the issue looks likely in kicad itself, not in the building/packaging of it, so I'm working ove at https://bugs.launchpad.net/kicad/+bug/1834717

yeswolf, 2019-10-07:
I can help to test, since I'm on Catalina and desperatly need a working Kicad :)

a7hybnj2, 2019-10-07:
@yeswolf since adamwolf just updated on the launchpad that he cannot get a build going I am going to guess its going to be a while.

I think I might upgrade to Catalina and just kicad on my other boxes for the time being.

adamwolf, 2019-10-07:
I don't know if that's entirely what I said :)

We can build on Catalina, everything I've tested with existing projects
works fine--but we cannot create new projects.  I've gotten a stack trace,
but it isn't clear enough what the issue is right now.  If it's something
in KiCad itself, I would bet we've got something figured out in just a few
days.  If it's upstream, it might be longer, depending on what it is and
how big the issue is.

a7hybnj2, 2019-10-07:
@adamwolf thanks for the clarification. Should we just be running the 10.14 nightlies or is there a catalina_beta somewhere?

adamwolf, 2019-10-07:
Right now, just run the 10.14 nightlies--but I'll scream from the
rooftops when we get it figured out :)

I seriously wonder if you just created a blank project and made copies
of that, if you could use Catalina today.  I don't think that's a
great workaround, but if the alternative is getting no work done...

a7hybnj2, 2019-10-07:
Lol, I already made a “kicad-empty” on my desktop to see if that works when I upgrade. I did not populate filetype though, will that be an issue. is the issue in creating new files? For instance does plotting gerbers work?

adamwolf, 2019-10-07:
I did not test plotting gerbers.  Let me know if it works :)

So far, the only issue I have seen is in creating a new project.

a7hybnj2, 2019-10-08:
So, I just created a new project without issue on catalina. When I opened kicad for the first time I gave permission to desktop and documents. Then I opened kicad which brought up my last project I had open and I hit 'new project' and I saved it which eneded up being in the documents directory.

There are some oddities like I when I click on desktop it says location doesn't exist (or something similar) but if I open a project that is stored on the desktop then hit new project I can save just fine...

adamwolf, 2019-10-08:
Interesting!  I was using the last public beta and it was much worse.

 I will reinstall my external drive soon !  (Could you possibly post on the
launchpad bug?)

a7hybnj2, 2019-10-08:
Yea, I will post there. But, it still isn't "right" it just is less "unusable" than I was lead to believe.

adamwolf, 2019-10-21:
I believe that creating new projects is working fine.  Let's move the discussion over to Launchpad for any other issues folks are having with Catalina.