KiCad/kicad-mac-builder #298: packaging is breaking the ngspice code models

lijon, 2019-09-30:
See https://bugs.launchpad.net/kicad/+bug/1835296

adamwolf, 2019-10-21:
lijon are you still seeing this?

lijon, 2019-10-27:
I just tried https://kicad-downloads.s3.cern.ch/osx/testing/5.1/kicad-5.1-nightly-20191026-103423-47deb8dea-10_14.dmg and it seems to work fine.

However, it's a bit confusing that libngspice is bundled at two locations in the app.
In lldb I could only see it referencing the one in Frameworks, but if I remove the other one (in PlugIns/sim/) the simulator fails to start:

13:05:26: dlopen(libngspice.dylib, 2): image not found
13:05:28: Unhandled exception class: St13runtime_error  what: Could not create simulator instance

adamwolf, 2019-10-27:
Thanks.   I'll close this.

We do not do this explicitly, it's something that comes with how
CMake's bundles things on macOS.  Feel free to look into it--I'd love
to understand why.