KiCad/kicad-mac-builder #301: Python is not compiled with https support

sethhillbrand, 2019-11-11:
Start the python console.  Type the following:

```Python
import urllib2
ret = urllib2.urlopen("https://www.google.com")
```

The return is
```Python
  File "/private/var/folders/h0/t3ynppn131jcvc9m8jcs38vr0000gn/T/AppTranslocation/5EA69F86-BCC4-4198-B6B4-8F54596E31B2/d/kicad.app/Contents/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib2.py", line 1266, in unknown_open
    raise URLError('unknown url type: %s' % type)
URLError: <urlopen error unknown url type: https>
```

Some googling leads to:
https://stackoverflow.com/questions/28376506/urllib-https-request-urlopen-error-unknown-url-type-https

> Application: KiCad
> Version: (5.1.5-rc1-13-g2997db8a2), debug build
> Libraries:
>     wxWidgets 3.0.4
>     libcurl/7.54.0 LibreSSL/2.6.4 zlib/1.2.11 nghttp2/1.24.1
> Platform: Mac OS X (Darwin 18.2.0 x86_64), 64 bit, Little endian, wxMac
> Build Info:
>     wxWidgets: 3.0.4 (wchar_t,STL containers,compatible with 2.8)
>     Boost: 1.69.0
>     OpenCASCADE Community Edition: 6.9.1
>     Curl: 7.54.0
>     Compiler: Clang 9.0.0 with C++ ABI 1002
> 
> Build settings:
>     USE_WX_GRAPHICS_CONTEXT=ON
>     USE_WX_OVERLAY=ON
>     KICAD_SCRIPTING=ON
>     KICAD_SCRIPTING_MODULES=ON
>     KICAD_SCRIPTING_PYTHON3=OFF
>     KICAD_SCRIPTING_WXPYTHON=ON
>     KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
>     KICAD_SCRIPTING_ACTION_MENU=ON
>     BUILD_GITHUB_PLUGIN=ON
>     KICAD_USE_OCE=ON
>     KICAD_USE_OCC=OFF
>     KICAD_SPICE=ON
>

adamwolf, 2019-11-14:
I'll put checking this in the python3 migration requirements.

adamwolf, 2019-11-14:
I think this may be a dupe of #226

sethhillbrand, 2019-11-15:
Oh bummer, it does appear the same.  I'll have to look at what the issue in #226 is to see if I can understand it.

We're going to need SSL support in v6 as we implement various internet-enabled plugins for manufacture and BOM lookup

adamwolf, 2019-11-15:
So for Python 2, it was really, really, really tricky to build a
Python 2 distribution that is relocatable *and* as a Framework.  I
ended up using a blog post from ‎Armin Ronacher, and IIRC he never got
the SSL stuff to work right.

There have been various advances in Python 3 for macOS that make it
easier, as far as I can tell, so I am hoping that this problem just
drops out when we get that going.

sethhillbrand, 2019-11-17:
After a bit of testing (I really needed the SSL working for a bit), I've found that if I copy the homebrew Python2 Framework over the KiCad framework, this all works.  I'm not sure what they do differently but we might be able to borrow it for Python3